OctoPrint/OctoPrint #4355: [Request] Uptime Octoprint

Oxize, 2021-12-27:
### Is your feature request related to a problem? Please describe.

No, but typing in constantly in SSH the uptime, when a simple fast uptime in System info page can do the same, would be nice to have.

### Describe the solution you'd like

Uptime feature under System info
[image]

### Describe alternatives you've considered

I can ssh into Octoprint, and use commands, but i think its a small code to code it into system info page.

tedder, 2021-12-29:
Are you wanting to see load or uptime? In any case it sounds like a great plugin.

Oxize, 2022-05-02:
?