OctoPrint/OctoPrint #4356: login dialog shows password failure behaviour if js is disabled

dothebart, 2021-12-28:
### The problem

1. Javascript in browser  (firefox) disabled (via noscript)
2. try to log into octoprint
3. the dialog behaves as if the user login wasn't successfull.

Expected behaviour:
UI should give feedback that it can't be used without javascript. This could be achieved by only activating the login button by javascript

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

0.18.0

### Operating system running OctoPrint

octopi

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2021-12-31:
The main UI has a `<noscript>` tag in it to display a message, I guess this needs adding to the login dialog too (and possibly the recovery page?). I think it can be done by 1.8.0.

foosel, 2022-01-17:
Indeed, that's the issue.

commit 00433095 (2022-01-17): 🥅 Add noscript to login page

Closes #4356