OctoPrint/OctoPrint #4359: [Discovery Plugin] mDNS responses send wrong IP out wrong interface

baconwaifu, 2022-01-03:
### The problem

Expected behavior: mDNS replies/advertisements contain an IP address on the associated interface
Actual (faulty) behavior: mDNS is advertising the IP from a completely different interface/subnet.

Details:
On a system with multiple IP-having interfaces (such as any system with docker installed, due to `docker0`) the internal mDNS stack (correctly) picks all interfaces and their associated IPs up. The problem lies in how it *uses* those IPs. I'm seeing it advertise the `docker0` IP out every interface it's connected to, in addition to that interface's IP.
This behavior is *normally* not problematic, as most systems with multiple interfaces are on networks that can talk between those two subnets, and even if they can't, they usually *know* they can't, and just use the other IPs. the real problem is host-internal networks that can exist on both hosts, but are entirely separate.

Reproduction: Install OctoPrint on a system that has docker installed, and try to run `avahi-resolve` against it a few times. you should see the `docker0` IP being advertised out the LAN.

(sidenote: the discovery plugin currently conflicts with avahi-daemon, suggest using that instead of internal if installed?)

Unfortunately the only way to fix this would be a pretty large overhaul to how mDNS is handled, with one registration per interface, or just going the easy workaround route and blacklisting `docker0` by default.

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

Issue is present from fresh install, safe mode irrelevant.

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

Raspberry Pi OS Bullseye

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Unsure why this report requires a SystemInfo for an obvious logic error, but here it is with redacted username: [octoprint-systeminfo-20220103203705.zip](https://github.com/OctoPrint/OctoPrint/files/7803969/octoprint-systeminfo-20220103203705.zip)
Yes I know the stock firmware on the enders is crap. it works well enough for now, until I can be arsed to upgrade it.

This issue persisted from the initial install, so the presence of 3rd party plugins is irrelevant. Noticed when chrome showed me `localhost` instead of the expected UI, as both my desktop and this pi have docker installed, with their respective `docker0` both having the same internal IP.

foosel, 2022-01-26:
Just to make sure we are talking about the same thing here -- are you trying to resolve the `.local` hostname (which isn't part of the Discovery plugin's job) or one of the services registered by the plugin, `_octoprint._tcp` or `_http._tcp`?

Does the issue persist if you limit the discovery plugin's registered interfaces, e.g. via 

``` yaml
plugins:
  discovery:
    interfaces:
      - eth0
      - wlan0
```

in `config.yaml`?

> the discovery plugin currently conflicts with avahi-daemon, suggest using that instead of internal if installed?

Conflicts in what way? From my understanding they both should just be sending and receiving messages on the multicast port. The discovery plugin uses the [zeroconf lib]() to do the zeroconf service registrations. AFAIK it's not possible to make that fall back to avahi backpacking (in fact it seems to one of its whole implementation feature to NOT depend on avahi) and since OctoPrint is cross-platform but avahi isn't, I also don't really see how to do this in a feasible and maintainable way. So if you could elaborate on the exact nature of the conflict you are observing in your environment (or give me precise reproduction steps and example outputs), that would be great to get to the bottom of this.

foosel, 2022-03-08:
@baconwaifu I'm still waiting on feedback from you here, otherwise I cannot look into this further.

dvoros, 2022-04-03:
Hi there @foosel. I'm running into the same problem.

It seems to me that the discovery plugin also responds to queries for `.local` addresses. I think this is what @baconwaifu could have referred to as "conflicting with avahi". It has also surprised me that Octoprint starts listening on `5353` even if avahi is already there, but apparently this is common practice when listening for multicast messages (see `SO_REUSEADDR`).

When trying to resolve the hostname Octoprint is running on (by `dig -p 5353 @224.0.0.251 dvoros-sajtos3.local`), I get responses coming from Octoprint. These responses include addresses on all interfaces:

[image]

I've confirmed this by enabling DEBUG logging for `zeroconf` (by doing `logging.getLogger('zeroconf').setLevel(logging.DEBUG)` in discovery plugin's `__init__`). This produced the output:

```log
2022-04-03 10:27:22,449 - zeroconf - DEBUG - Received from '192.168.0.139':46959 [socket 4 (('0.0.0.0', 5353))]: <DNSIncoming:{id=55981, flags=288, truncated=False, n_q=1, n_ans=0, n_auth=0, n_add=1, questions=[a[question,QM,in,dvoros-sajtos3.local.]], answers=[]}> (61 bytes) as [b'\xda\xad\x01 \x00\x01\x00\x00\x00\x00\x00\x01\x0edvoros-sajtos3\x05local\x00\x00\x01\x00\x01\x00\x00)\x10\x00\x00\x00\x00\x00\x00\x0c\x00\n\x00\x08\xe7\xe0\x84]\xe7\x8f3\x8d']
...
2022-04-03 10:27:22,450 - zeroconf - DEBUG - Sending to (192.168.0.139, 46959) (134 bytes #1) <DNSOutgoing:{multicast=False, flags=33792, questions=[a[question,QM,in,dvoros-sajtos3.local.]], answers=[(record[a,in-unique,dvoros-sajtos3.local.]=120/119,172.18.0.1, 0), (record[a,in-unique,dvoros-sajtos3.local.]=120/119,192.168.0.139, 0), (record[a,in-unique,dvoros-sajtos3.local.]=120/119,172.22.0.1, 0), (record[a,in-unique,dvoros-sajtos3.local.]=120/119,192.168.49.1, 0), (record[a,in-unique,dvoros-sajtos3.local.]=120/119,172.19.0.1, 0), (record[a,in-unique,dvoros-sajtos3.local.]=120/119,172.17.0.1, 0)], authorities=[], additionals=[]}> as ...
```

I've checked this on current master (`76e87ba81`). Disabling the discovery plugin obviously resolves the issue.

foosel, 2022-04-04:
@dvoros Could you check if the new options introduced by [2e266b5](https://github.com/OctoPrint/OctoPrint/commit/2e266b5af5deb9e35e1c92fd86cc59b337df9ffe) and available in 1.8.0rc3 (maintenance release channel) are working to resolve this for you?

`plugins.discovery.ignoredAddresses` and `plugins.discovery.ignoredInterfaces` now both take a list of addresses respectively interface names to ignore and not send discovery announcements for. The interfaces get used for initializing the zeroconf stack and should hopefully work as advertised (they did so in my tests, but as usual, more tests are better). Alternatively, setting `plugins.discovery.addresses` to only the addresses for which announcements *should* be done should work as well now.

foosel, 2022-06-20:
@baconwaifu @dvoros could either of you please confirm or deny this is still an issue with the aforementioned new options available as of 1.8.0?

closed 2022-08-08 by foosel