OctoPrint/OctoPrint #4360: [Request]  CR-10 v2 Firmware 1.70 BL Filament Detector Support

HaToan, 2022-01-05:
### Is your feature request related to a problem? Please describe.

OctoPrint 1.7 does not support the built-in filament breakage/end detector for CR10s-pro v2.

### Describe the solution you'd like

I capture serial.log when I printing from SDcard and filament breakage as follows:
```
Send: M27
Recv: SD printing byte 43642/3915639
Recv: ok
Recv:  == T:199.80 /200.00 == B:49.92 /50.00 @:73 B@:0
Send: M27
Recv: SD printing byte 44140/3915639
Recv: ok
Send: M27
Recv: echo:enqueueing "G28 X0 Y0"
Recv:  == T:199.89 /200.00 == B:49.74 /0.00 @:71 B@:0
Recv: echo:busy: processing
Recv:  == T:199.86 /200.00 == B:49.72 /0.00 @:72 B@:0
Recv: echo:busy: processing
Recv:  == T:199.94 /200.00 == B:49.49 /0.00 @:71 B@:0
Recv: echo:busy: processing
Recv:  == T:200.00 /200.00 == B:49.41 /0.00 @:70 B@:0
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:5.00 E:45.14 Count X:0 Y:0 Z:2000
Recv: SD printing byte 44511/3915639
Recv: ok
Send: M27
Recv: SD printing byte 44511/3915639
Recv: ok
Recv:  == T:200.00 /0.00 == B:49.29 /0.00 @:0 B@:0
Send: M27
Recv: SD printing byte 44511/3915639
Recv: ok
```
when I press button "Yes" on screen to continue printing 
```
Send: M27
Recv: SD printing byte 69291/3915639
Recv: ok
Recv:  == T:200.00 /200.00 == B:45.43 /0.00 @:78 B@:0
Recv: echo:enqueueing "M190 S50"
Recv: echo:enqueueing "M109 S200"
Recv: echo:enqueueing "G0 Z0.80"
Recv: echo:enqueueing "G92 E50.14"
Recv:  == T:200.00 /200.00 == B:45.41 /50.00 @:77 B@:0 W:?
Send: M27
Recv:  == T:200.00 /200.00 == B:45.31 /50.00 @:76 B@:127 W:?
Recv:  == T:199.97 /200.00 == B:45.33 /0.00 @:76 B@:127
Recv:  == T:199.97 /200.00 == B:45.33 /0.00 @:76 B@:127 W:?
Recv:  == T:199.91 /200.00 == B:45.51 /0.00 @:76 B@:127 W:9
Recv: echo:busy: processing
Recv:  == T:199.86 /200.00 == B:45.53 /0.00 @:77 B@:127
Recv:  == T:199.86 /200.00 == B:45.53 /0.00 @:77 B@:127 W:7
Recv:  == T:199.89 /200.00 == B:45.71 /0.00 @:76 B@:127 W:6
Recv: echo:busy: processing
Recv:  == T:199.89 /200.00 == B:45.82 /0.00 @:76 B@:127
Recv:  == T:199.89 /200.00 == B:45.82 /0.00 @:76 B@:127 W:5
Recv:  == T:199.89 /200.00 == B:46.05 /0.00 @:75 B@:0 W:4
Recv: echo:busy: processing
Recv:  == T:199.91 /200.00 == B:46.20 /0.00 @:75 B@:0
Recv:  == T:199.91 /200.00 == B:46.20 /0.00 @:75 B@:0 W:3
Recv:  == T:199.97 /200.00 == B:46.32 /0.00 @:73 B@:0 W:2
Recv: X:0.00 Y:0.00 Z:0.80 E:50.14 Count X:0 Y:0 Z:1987
Recv: SD printing byte 69381/3915639
Recv: ok
Send: M27
Recv: SD printing byte 69620/3915639
Recv: ok
Send: M27
Recv:  == T:200.00 /200.00 == B:46.40 /0.00 @:73 B@:0
Recv: echo:busy: processing
```

jneilliii, 2022-01-05:
I'm pretty sure this is too printer/firmware specific to support on a global scale, so a plugin would be the best approach for supporting this. The problem here is that the firmware is not telling the host software (OctoPrint in this case) of when the runout happens in any usable way. OctoPrint completely supports host_action_commands and host_prompt_support, which is the proper way for printers to interact with the connected hosts. There are pre-built Marlin binaries around from several different sources that enable these options for CR-10, a couple that come to mind are jyers (I think) and InsanityAutomation.

cp2004, 2022-01-05:
There's nothing in that serial.log to indicate that the filament has run out and the print should be paused, so OctoPrint can't support it. You'll need a firmware that works properly which usually means one that doesn't come from Creality.

closed 2022-01-05 by cp2004