OctoPrint/OctoPrint #4362: GitHub tags are only read up to the first `-` character, breaking tags such as `1.0.0-rc1`

cp2004, 2022-01-05:
### The problem

The software update plugin does a load of parsing to compare the GitHub tag version to the one from the installed Python packages. I unwrapped some of it to figure out what was going on here.

We have a tag on GitHub with the name `1.0.0-rc1`, which according to [PEP440](https://www.python.org/dev/peps/pep-0440/#pre-release-separators) is allowed but would be normalized to `1.0.0rc1`, which is how it comes through if this tag is set in setup.py.

The plugin runs it through (as far as I could trace) `octoprint.plugins.softwareupdate.version_checks.github_release.get_comparable_version`. This calls some functions in `octoprint.util.version` including `normalize_version`, which only reads the tag up to and including the `-`, stripping it down to just `1.0.0` in this case, which would be seen as 'ahead' of `1.0.0rc1` of course.

Some testing to demonstrate, the first line is the problem one:
```python
>>> get_comparable_version("1.1.1-rc1")
<Version('1.1.1')>
>>> get_comparable_version("1.1.1rc1")
<Version('1.1.1rc1')>
>>> get_comparable_version("1.1.1.rc1")
<Version('1.1.1rc1')>
>>> get_comparable_version("1.1.1_rc1")
<Version('1.1.1rc1')>
```

They should all be parsed equally according to PEP440.

This resulted in an update loop where `1.0.0-rc1` was set the same in both `setup.py` and the GitHub tag, but it became GitHub tag = 1.0.0 and setup.py = 1.0.0rc1 when parsed and the comparison would always show an update available.

**TLDR**; everything after the `-` in a GitHub tag is ignored, when it should be included.

Discussion on discord: https://discord.com/channels/704958479194128507/708230829050036236/928402397435420775

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

NA

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

All

### Printer model & used firmware incl. version

All

### Browser and version of browser, operating system running browser

All

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

synman, 2022-01-06:
I love bots looking out of us :).  Hopefully they don't murder all of us post singularity.

The discussion the bot tagged here is for a different issue.   @cp2004 is addressing an issue specific to tag parsing.  the issue covered in the tagged discussion is tied to the `github_commit` release type not properly applying pip args.  Basically, octoprint needs to tell pip to force a reinstall with this strategy rather than comparing version information.

foosel, 2022-01-17:
I'm pretty sure this "only up until -" thing stems from a former OctoPrint versioning scheme that has not been used for half a decade now, so it should be an easy fix really.

foosel, 2022-02-02:
Fix by @GonzoDMX merged and to be pushed out with 1.8.0 🍻

cp2004, 2022-05-17:
1.8.0 has been released 🎉

PR #4389 (2022-02-02): Addresses issue #4362
#### What does this PR do and why is it necessary?
Fixes the normalize_version function in src/util/version.py so that it returns the correct version when the input version string is a release candidate containing the characters "-rc".

#### How was it tested? How can it be tested by the reviewer?
Tested in OctoPrint dev environment.

#### What are the relevant tickets if any?
Fixes #4362