OctoPrint/OctoPrint #4363: github_commit release type incorrectly performs version check

synman, 2022-01-06:
### The problem

Using the following software update strategy, weird things happen if you do not include a version update for your plugin with any commits you push to the identified origin repo/branch.

    def get_update_information(self):
        # Define the configuration for your plugin to use with the Software Update
        # Plugin here. See https://github.com/foosel/OctoPrint/wiki/Plugin:-Software-Update
        # for details.

        useDevChannel = self._settings.get_boolean(["useDevChannel"])

        if useDevChannel:
            return dict(bettergrblsupport=dict(  # version check: github repository
                                                 # update method: pip
                displayName='Better Grbl Support (Development Branch)',
                type='github_commit',
                user='synman',
                repo='OctoPrint-Bettergrblsupport',
                branch="devel",
                current="fd0b1bac7a23ba4b01f58353c7a19c6bc4ea219e",
                method="pip",
                pip='https://github.com/synman/Octoprint-Bettergrblsupport/archive/refs/heads/devel.zip'))

        else:

Since the version check indicates no change, pip does not do an actual re-installation of the package:

```
2022-01-05 19:21:42,181   /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/synman/Octoprint-Bettergrblsupport/archive/2d4438163da8627d86f80b3236189df32d45fd89.zip --no-cache-dir
2022-01-05 19:21:44,295 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2022-01-05 19:21:44,296 > Collecting https://github.com/synman/Octoprint-Bettergrblsupport/archive/2d4438163da8627d86f80b3236189df32d45fd89.zip
2022-01-05 19:21:44,297 > Downloading https://github.com/synman/Octoprint-Bettergrblsupport/archive/2d4438163da8627d86f80b3236189df32d45fd89.zip
2022-01-05 19:21:46,433 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Better-Grbl-Support===2.1.0-rc1.3) (1.7.2)
2022-01-05 19:21:46,435 > Requirement already satisfied: feedparser<7,>=6.0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (6.0.8)
2022-01-05 19:21:46,435 > Requirement already satisfied: psutil<6,>=5.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (5.8.0)
2022-01-05 19:21:46,436 > Requirement already satisfied: zeroconf<0.34,>=0.33 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.33.4)
2022-01-05 19:21:46,436 > Requirement already satisfied: PyYAML<6,>=5.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (5.4.1)
2022-01-05 19:21:46,436 > Requirement already satisfied: Jinja2<3,>=2.11.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2.11.3)
2022-01-05 19:21:46,436 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (3.1.1)
2022-01-05 19:21:46,436 > Requirement already satisfied: OctoPrint-PiSupport>=2021.10.28 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2021.10.28)
2022-01-05 19:21:46,436 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (3.5)
2022-01-05 19:21:46,437 > Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2.8.5)
2022-01-05 19:21:46,437 > Requirement already satisfied: pkginfo<2,>=1.7.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.7.1)
2022-01-05 19:21:46,437 > Requirement already satisfied: flask<2,>=1.1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.1.4)
2022-01-05 19:21:46,437 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2021.10.11)
2022-01-05 19:21:46,437 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.4)
2022-01-05 19:21:46,437 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.18.2)
2022-01-05 19:21:46,437 > Requirement already satisfied: pathvalidate<3,>=2.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2.5.0)
2022-01-05 19:21:46,438 > Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2021.2.23)
2022-01-05 19:21:46,438 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2.0)
2022-01-05 19:21:46,438 > Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.11.0)
2022-01-05 19:21:46,438 > Requirement already satisfied: wrapt<1.13,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.12.1)
2022-01-05 19:21:46,438 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.1.1)
2022-01-05 19:21:46,438 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.5.0)
2022-01-05 19:21:46,438 > Requirement already satisfied: zipstream-new<1.2,>=1.1.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.1.8)
2022-01-05 19:21:46,439 > Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.8.0)
2022-01-05 19:21:46,439 > Requirement already satisfied: sarge==0.1.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.1.6)
2022-01-05 19:21:46,439 > Requirement already satisfied: immutabledict<3,>=2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (2.2.1)
2022-01-05 19:21:46,439 > Requirement already satisfied: websocket-client<1,>=0.59 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.59.0)
2022-01-05 19:21:46,439 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.2.0)
2022-01-05 19:21:46,439 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (7.1.2)
2022-01-05 19:21:46,439 > Requirement already satisfied: cachelib<0.2,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.1.1)
2022-01-05 19:21:46,440 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.0.1)
2022-01-05 19:21:46,440 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.0.7)
2022-01-05 19:21:46,440 > Requirement already satisfied: colorlog<6,>=5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (5.0.1)
2022-01-05 19:21:46,440 > Requirement already satisfied: emoji<2,>=1.4.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.6.1)
2022-01-05 19:21:46,440 > Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (0.10.4)
2022-01-05 19:21:46,440 > Requirement already satisfied: sentry-sdk<2,>=1.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Better-Grbl-Support===2.1.0-rc1.3) (1.4.3)
[... cut ...]
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220106105117.zip](https://github.com/OctoPrint/OctoPrint/files/7823030/octoprint-systeminfo-20220106105117.zip)

NilsRo, 2022-01-07:
--force-reinstall worked for me

Perhaps this can be added as a new parameter for the updater to enable this if needed and still using standard pip implementation.

cp2004, 2022-01-07:
As I mentioned on discord when we were figuring it out, OctoPrint only automatically applies these extra arguments if:
1. The line starts with 'Requirement already satisfied (use --upgrade to upgrade);
2. The line contains the 'target', which in this case would end up as `https://github.com/synman/OctoPrint-Bettergrblsupport/archive/*********************.zip` in this case.

Neither of which show up in pip's output above, which is generally unhelpful.

> NOTE: --no-deps may be too restrictive.

With this, it might be restrictive (if your dependencies change) *however*, it would be more problematic to leave it out. If you do a force-reinstall without it, it will *also* reinstall every dependency, which will include OctoPrint & all it's dependencies as well, which will take a while.

A note on the `--ignore-installed` - it *might* be necessary in some situations. Pip's docs say it just overwrites whatever packages are present, because pip is not the only package manager (though by far the most used) it could break something there, but otherwise it doesn't even care about what's already installed & just overwrites it. When suggesting the arguments, I took note of how OctoPrint already used it.

foosel, 2022-01-24:
This is a tricky issue. 

The way things work is this:
  1. OctoPrint installs the target with a more or less regular `pip install`
  2. If the target gets reported as already installed ("requirement satisfied"), OctoPrint triggers a second `pip install`, this time with `--force-reinstall --ignore-installed --no-deps`

Here, step 2 never triggers due to pip not complaining. As @cp2004 said, not helpful, and a behaviour that needs investigation since it might be "regression" (from our perspective, not pip's) that needs a workaround in OctoPrint.

I can't just always `--force-reinstall`, as this would reinstall all deps as well, as mentioned by @cp2004 - that's a no-go as it takes WAY too long. The two-tiered approach is done because of there not being a better way (at least that I'm aware of, happy to be corrected here) to tell pip to reinstall the package in question, but *only* the package in question, not any of its dependencies, unless required by the target. However, I'm going to add a `force_reinstall` setting you can set to true in the update config and which will enforce the two tiered run, even if pip doesn't complain.

foosel, 2022-01-26:
So, two changes that will be in 1.8.0:

  1. The aforementioned `force_reinstall` flag you can set on your update config
  2. a different "already installed" logic, now no longer depending on "Requirement already satisfied" plus install URL, but instead "Requirement already satisfied" plus a *lack* of a "Successfully installed" line. That should work in most cases.

synman, 2022-02-01:
thank you for giving us a path forward with this!

commit 84a049ba (2022-01-26): ♻️ Use centralized pip utils for SWU & PMGR

We don't want to reimplement already-installed-,
egg-problem- and other logic in multiple places,
so centralize this in octoprint.util.pip and
also add a test suite.

already-installed-logic has also been updated
to solve #4363 by now checking if there's been
a successful result line and if not if there's
been an "already satisfied" line, which should
work more universally across pip versions.

Closes #4363