OctoPrint/OctoPrint #4368: [Request]Make configurable the g-code sent when STOP button is pressed and a SD print is on going

mriscoc, 2022-01-13:
### Is your feature request related to a problem? Please describe.

When the printer (Ender3v2) is printing from the SD card and STOP button is pressed in the octopi console, the Octoprint is issuing a M25 command:
```
Changing monitoring state from "Printing from SD" to "Cancelling"
Send: M108
Send: M25
```
That starts the pause sequence in the printer, if Advanced Pause is active, the printer goes to the parking position and wait for the user to continue the print, it is not possible to cancel the pause.

There is a better command for current Marlin 2.x printers: M524 https://marlinfw.org/docs/gcode/M524.html this command allows cancel the print and return the control to the user.

### Describe the solution you'd like

The g-code for cancel a SD print could be configurable, put a M524 in the "After print job is cancelled" g-code script do not prevent the M25 to be send.

Octoprint could check if the printer firmware is Marlin 2.x and make the change automatically

jneilliii, 2022-01-13:
It does seem that this is hard coded [here](https://github.com/OctoPrint/OctoPrint/blob/53eb9bf979246db739c4326fda1f186b91840554/src/octoprint/util/comm.py#L1755) (maintenance branch). It would make sense for this to be configurable.

Dominik-1980, 2022-02-07:
I can confirm if I print from SD on my Anycubic Vyper, I can't pause or cancel the print in octoprint.
I stucks in "canceling" forever and the only way is to turn the printer off and on again.
The M524 command Is what we need on Marlin 2.x
If I send it in the terminal the printer cancels the print and shows me the message in the touchscreen.

mriscoc, 2022-02-07:
For more recent printers it could be helpful that the default value was M524, and configurable to M25 if it is needed.

foosel, 2022-02-08:
@jneilliii I see that you made a commit referencing this issue in your fork - do you plan on sending a PR for this, are there issues I can help ironing out so we can get this included in 1.8.0?

jneilliii, 2022-02-08:
My issue was in the commit note. Not sure why it's not initially binding. I'll start a PR so we can discuss.

jneilliii, 2022-02-09:
@mriscoc the changes have been merged with the above commit and will be available as part of version 1.8.0.

mriscoc, 2022-02-09:
Excellent, thank you.

cp2004, 2022-05-17:
1.8.0 has been released 🎉

closed 2022-05-17 by cp2004