OctoPrint/OctoPrint #4369: [Request] Add more COLORS to navbar!

bolsoncerrado, 2022-01-13:
### Is your feature request related to a problem? Please describe.

Hi!

 I run a farm where every printer and octopi instance is a different color and I've already ran out of choices! :P

THanks!

### Describe the solution you'd like

Would it be possible to add a text field next to "Color:" dropdown in Settings -> Appearance so we can input RGB codes for the navbar appearance?

### Describe alternatives you've considered

A plugin maybe?

### Additional context

[image]

arroyoDev, 2022-01-19:
@bolsoncerrado how do you get the appearance settings to stay?  I run several instances of octoprint on the same machine and any change to one overrides the appearance of all the others. 

Does this not happen to you?

jneilliii, 2022-01-19:
It shouldn't overwrite the others if you are using separate basedir settings for each instance (which you should do).

bolsoncerrado, 2022-01-19:
I run every octoprint with its dedicated pi anyway.

deepsol-oba, 2022-05-28:
@bolsoncerrado - Have you considered using a plugin like "Themify". It will allow you to modify some of the UI experience as requested by you.

bolsoncerrado, 2022-05-28:
Might have to try...although I think my proposed solution was far easier.... A simple RGB field the UI could call to.

closed 2022-06-17 by bolsoncerrado