OctoPrint/OctoPrint #4371: [Request] File Move event

OllisGit, 2022-01-16:
### Is your feature request related to a problem? Please describe.

My PrintJobHistory-Plugin stores the printed filename (with path) in the database and provides the user the functionality for "re-selecting" this file. 
The file must exists in the filesystem (gcode is not stored in database or different folder).

Unfortunately the user (or on other plugin like "DeleteMoveAfterPrint") can move the folder to an sub-folder.
Currently I receive two events "remove/add", but to modify the database entry I need both informations at once:
- look for printjob which used this gcode
- update the selected printjob with the new location

Do you plan to add this event?

### Describe the solution you'd like

Fire a new additional event.
Beside fireing remove/added-event, fire also Events.FILE_MOVED, I need something like this in ```move_file```:
```
        eventManager().fire(
            Events.FILE_REMOVED,
            {
                "storage": destination,
                "path": source_path_in_storage,
                "name": source_name,
                "type": get_file_type(source_name),
            },
        )
        eventManager().fire(
            Events.FILE_ADDED,
            {
                "storage": destination,
                "path": dst_path_in_storage,
                "name": dst_name,
                "type": get_file_type(dst_name),
            },
        )

        eventManager().fire(
            Events.FILE_MOVED,
            {
                "storage": destination,
                "source_path": source_dst_path_in_storage,
                "target_path": target_dst_path_in_storage,
                "name": dst_name,
                "type": get_file_type(dst_name),
            },
        )

```

### Describe alternatives you've considered

My current "Gedankenspiel" is to hijack the JavaScript-Move-Operation or overwrite the move_file function from filemanger...not a good one ;-)

jneilliii, 2022-01-16:
This feels like a duplicate... searching.

jneilliii, 2022-01-16:
Duplicate #3491

closed 2022-01-16 by jneilliii