OctoPrint/OctoPrint #4373: can we implement a way of exporting/importing plugins from one instance of octoprint to another?

r0dnay, 2022-01-18:
### Is your feature request related to a problem? Please describe.

its a tedious task to reinstall all plugins present on one instance, on to another instance

### Describe the solution you'd like

an import and export button in plugin manager

an export button exists but just creates a .json file with paths and names of plugins and puts it in the default download folder

foosel, 2022-01-18:
I also can't shake the feeling that this is a duplicate. I'm VERY sure I saw this request before because I even remember thinking about how best to implement it based on the plugin export.

jneilliii, 2022-01-18:
I thought so too, but couldn't find it with quick search.

cp2004, 2022-01-18:
The idea was probably discussed when I implemented the export button, but I never got on to it and no one else has either yet.

r0dnay, 2022-01-18:
> I created a python script for someone on Discord to do this from a file extracted from a backup on Windows, since you can't restore a backup on Windows. It would also work for a pi, but would require changing the path to a file in the script. I agree this is not ideal as it is not part of the UI, but at least gets you closer. SSH to the pi, activate the octoprint venv (`source /home/pi/oprint/bin/activate`) and then run `python` and copy paste the content of the file linked below (replacing file path to plugin_list.json extracted from a backup).
> 
> https://gist.github.com/jneilliii/61478a04a1cdaa96ae89549b46de27b8

thanks, in the venv at the prompt i run python from the prompt?
thats what i did, and it gives >>> that prompt, then i pasted the code from your link (changed the path to the json file download location) but im assuming due to the lack of formatting, it presents a syntax error.

is there another way? such as pasting it in visual code/studio and saving the file or something?

r0dnay, 2022-01-18:
> The idea was probably discussed when I implemented the export button, but I never got on to it and no one else has either yet.

what was the initial motive for the export button? i mean it outputs the text so its good to see at a glance which plugins you have, just seems a little dsiconnected, is there a use for it currently?

thanks for the input people, much appreciated

cp2004, 2022-01-18:
Try `curl -L https://gist.github.com/jneilliii/61478a04a1cdaa96ae89549b46de27b8/raw/2736975eba6664c0539de305822923823199162f/windows_install_plugins_from_backup.py --output install_plugins_from_backup.py` to download the file, then you should have a file saved, you can edit it with `nano install_plugins_from_backup.py` to change the path, and you can run `python install_plugins_from_backup.py` (with the venv activated). Or something like that.

> what was the initial motive for the export button? i mean it outputs the text so its good to see at a glance which plugins you have, just seems a little dsiconnected, is there a use for it currently?

I believe it was a requested feature, for an export list. I would agree that it is not much use without a relevant import button as well, maybe people use it for scripts like the one above. I didn't have time to implement the import button at the same time so just put in the PR with what I had.

r0dnay, 2022-01-18:
ah ok, ive got the file and edited the path, just struggling with the last bit  `python install_plugins_from_backup.py` 

does the install_plugins_from_backup.py file need to be in a specific directory?

from the >>> prompt i get a syntax error when i type the command like that

i tried the full path too
C:\Marlin-bugfix-2.0.x\install_plugins_from_backup.py

> I believe it was a requested feature, for an export list. I would agree that it is not much use without a relevant import button as well, maybe people use it for scripts like the one above. I didn't have time to implement the import button at the same time so just put in the PR with what I had.

well if you hadnt, it wouldnt have occured to me when i embarked on the task of re downloading plugins, that import export would be good to back up and restore settings/plugins etc so good thing you left it in there

jneilliii, 2022-01-18:
You wouldn't enter that from the python prompt `>>` you would just run that at the command prompt with venv activated.

r0dnay, 2022-01-18:
sorry about that :s 

i have the edited `install_plugins_from_backup.py` and save it in `C:\Octoprint\venv\Scripts>`

this is what i get

```
(venv) C:\Octoprint\venv\Scripts>python install_plugins_from_backup.py
Traceback (most recent call last):
  File "C:\Octoprint\venv\Scripts\install_plugins_from_backup.py", line 12, in <module>
    with open(plugin_list) as file:
OSError: [Errno 22] Invalid argument: '\u202ag:\\Downloads\\plugin_list.json'

(venv) C:\Octoprint\venv\Scripts>
```

any idea why the path g:\downloads\plugin_list.json appears like that?

r0dnay, 2022-01-18:
actually ive just changed the path in the .py file to 

`plugin_list = (r"‪g:\Downloads\plugin_list.json","r")`

now it comes up with

`(venv) C:\Octoprint\venv\Scripts>python install_plugins_from_backup.py
Traceback (most recent call last):
  File "C:\Octoprint\venv\Scripts\install_plugins_from_backup.py", line 12, in <module>
    with open(plugin_list) as file:
TypeError: expected str, bytes or os.PathLike object, not tuple`

cp2004, 2022-01-18:
Well that's because you changed the path to be a tuple, which is not what it wants, it is looking for a simple string like it was earlier, and is now complaining about that change.

This conversation would probably be easier to have on the community forums or discord server, to debug this particular script, leaving this discussion for that around actually implementing something like this.

r0dnay, 2022-01-18:
ah ok, i thought if i just change the path in the .py file and run it as advised itd work, think im having issues with syntax, cant seem to get it to accept the `python install_plugins_from_backup.py` command

initially the only thing i changed in the script was the path
 `# path to file extracted from backup
plugin_list = "‪g:\Downloads\plugin_list.json"`

do you have a link to the community forum/discord?

thanks for all your help - much appreciated

cp2004, 2022-01-18:
Discord server - https://discord.octoprint.org  (probably fastest way to get help with it)
Community forums https://community.octoprint.org

r0dnay, 2022-01-18:
great thanks, can i link this thread in a post on both those sites?

cp2004, 2022-06-20:
That was my idea - to pair it up with the export button I implemented. It is on my list.

foosel, 2022-11-17:
@cp2004 still on your list? Otherwise I'll take a shot at this next week so we can still include it in 1.9.0 :)

cp2004, 2022-11-17:
I would love to do it, but I know I won't be able to do it so go for it if we can get it into 1.9.0. It never really got past a concept stage for me, to do some validation of the json and then install the plugins.

foosel, 2023-01-18:
Implemented and ready for 1.9.0.

In the end I had to implement a different solution than to simply reuse the install queue, since that actual installation process is handled client side whereas I wanted the ability to also install plugins from exports unattended and headless. Long term it will probably make sense to see if we can maybe merge both features somehow, but for now this should do.

Funnily enough this request was opened precisely a year ago ^^

foosel, 2023-05-23:
1.9.0 has been released

closed 2023-05-23 by foosel