OctoPrint/OctoPrint #4380: Connectivity check does not take IPv6 addresses

wtremmel, 2022-01-22:
### The problem

My OctoPI runs in an IPv6-only network (IPv4 connectivity to the outside is done via NAT64 and DNS64).

When I try to enter an IPv6 address into the "Host IP" field it is accepted (I can save it) but the check for connectivity always fails.

I tried:
- 2620:fe::fe, port 53
- 2001:4860:4860::8888, port 53
- using "host octoprint.org 2620:fe::fe" from the command line works
- "Test name resolution" also works

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Not relevant (but Ender 3v2)

### Browser and version of browser, operating system running browser

Safari

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220122202638.zip](https://github.com/OctoPrint/OctoPrint/files/7919120/octoprint-systeminfo-20220122202638.zip)

cp2004, 2022-01-22:
Bug confirmed, my initial response linked below - something extra likely needed in the code to handle IPv6.

https://community.octoprint.org/t/connectivity-check-in-an-ipv6-only-network/41034/2?u=charlie_powell

cp2004, 2022-01-22:
It may be possible to change this conditionally:
https://github.com/OctoPrint/OctoPrint/blob/5a1b07e46f92ee3448ff99d36f1ab1ecefb62c1c/src/octoprint/util/net.py#L229-L231

Instead of `socket.AF_INET` using `socket.AF_INET6`.

wtremmel, 2022-01-23:
Why not have a field where you can enter a host name and then do the Happy Eyeballs thing? (try pinging both IPv4 and IPv6 and if one if successful there is reachability)

foosel, 2022-01-24:
Happy Eyeballs it is. If IPv6 support is detected, `server_reachable` will now first try to reach the host via a v6 socket and if that doesn't work fall back to v4. If that doesn't work either, it'll report failure. Will be rolled out with 1.8.0.

commit 33dbdac0 (2022-01-24): 🐛 Fix connectivity check against IPv6 hosts

Closes #4380