OctoPrint/OctoPrint #4381: Newest octoprint 1.7.3 now pause print is not starting the z back in the right spot

12345Josparky, 2022-01-24:
Just updated to the newest octoprint 1.7.3 and now my printers are not starting back at the correct spot on the z. It is starting too low on the z now. This only started after updating to this latest version. Can anyone help with this please. Or can I go back a version this is not good for me at all. I need to get some prints done and this is messing me up.

jneilliii, 2022-01-24:
OctoPrint on its own does not change anything in regard to your printer or the gcode commands that are sent to your printer. I don't see anything in your logs (serial log is disabled) and no indication that you attempted in safe mode as requested by the bug template, which would rule out possible plugin related conflicts. 

To downgrade to the previous version you can use the zip url from the releases here on GitHub in plugin manager > get more > ...from URL.

12345Josparky, 2022-01-24:
Thank you for replying I am not sure exactly what happened but it was working fine until this update then it all started. I am not understanding what you mean about how to go back to the version I had before to see if that helps. Can you give little more direction please.

12345Josparky, 2022-01-24:
So are you saying I just put this url in under the from url and hit install to go back to this older version?? https://github.com/OctoPrint/OctoPrint/releases/tag/1.7.2

12345Josparky, 2022-01-24:
I tried that and got this. I am no expert at this at all

Installing plugin from https://github.com/OctoPrint/OctoPrint/releases/tag/1.7.2...
Error!
Could not install plugin from https://github.com/OctoPrint/OctoPrint/releases/tag/1.7.2, was neither a plugin archive nor a single file plugin

jneilliii, 2022-01-24:
no, the url for the zip file asset attached to that, so this...I don't think it's going to change anything for you though because your issue sounds like relative vs absolute positioning possibly, 

```
https://github.com/OctoPrint/OctoPrint/archive/refs/tags/1.7.2.zip
```

jneilliii, 2022-01-24:
if you want to debug the situation you should reach out in the Discord server's #support-printer channel.

12345Josparky, 2022-01-24:
Thank you I will give it a try. All I know is been working for months and I haven't changed anything else but this so not sure why it started doing this but updated both my printers with this new version and they are both doing the exact same thing now that has been working until now.

12345Josparky, 2022-01-24:
Didn't know there was a discord channel is it a texting channel only or talking too

jneilliii, 2022-01-24:
there are audio channels, but not too many people jump in those too often.

cp2004, 2022-01-24:
If you are going to report a bug, then fill in the [Bug report form](https://github.com/OctoPrint/OctoPrint/issues/new?assignees=&labels=&template=bug_report.yml) as that provides the necessary information, rather than a short description that is not enough to work with here.

12345Josparky, 2022-01-25:
sorry cp2004 I am all new to this code stuff man I will gladly do whatever is needed. right now testing the olser version I rolled it back to to see if it was causing a conflict with a plugin but jneilliii is probably correct in that it may be something in the gcode. but I haven't changed anything playing around with all that now and doing test prints to try and figure it out. Sorry for not doing things in the way the need to be done.

12345Josparky, 2022-01-25:
> 

I guess I am pretty dumb I can't find that channel on discord. put it in the search and nothing comes up

jneilliii, 2022-01-25:
https://discord.com/channels/704958479194128507/708230166232694855

foosel, 2022-01-25:
There were no changes made to anything related to printer communication, which [the changelog for 1.7.3 also quickly reveals](https://github.com/OctoPrint/OctoPrint/releases/tag/1.7.3), and this isn't a proper bug report in any case. As already pointed out, it makes more sense to tackle this as a support request, and for this there are different places, the aforementioned Discord being one of them, [the Community Forums](https://community.octoprint.org/c/support/7) being another. In either case though, you'll have to provide at the very least a [system info bundle with enabled serial.log]() or it will be extremely hard to assist you.

Since there is no immediate development TODO here, I'm closing this now, should it turn out there's a bug here after all over the course of support we can always reopen it.

closed 2022-01-25 by foosel