OctoPrint/OctoPrint #4392: Octoprint failed to update to pip 22.0.2

jcwren, 2022-01-31:
### The problem

Octoprint has been running flawlessly for months, including regular updates. Opened Octoprint, was notified of an update available, told it to start.  Update failed with the following message in the log file.

```
2022-01-31 14:51:58,398 - octoprint.plugins.PrintJobHistory - INFO - Plugin-State:
| PreHeat=enabled (0.8.0)
| filamentmanager=enabled (1.9.1)
| DisplayLayerProgress=enabled (1.27.2)
| SpoolManager=enabled (1.6.0)
| UltimakerFormat=enabled (1.0.1)
| PrusaSlicerThumbnail=enabled (1.0.0)
| costestimation=missing (None)

2022-01-31 14:51:58,403 - octoprint.plugins.PrintJobHistory - WARNING - Filamentracking is disabled, but some plugins are installed!
2022-01-31 14:51:59,525 - octoprint.server.util.sockjs - INFO - User jcw logged in on the socket from client ::ffff:172.16.0.10
2022-01-31 14:52:01,883 - octoprint.plugins.tracking - INFO - Sent tracking event webui_load, payload: {'browser_name': 'Chrome', 'browser_version': '97.0.4692.99', 'os_name': 'Windows', 'os_version': '7'}
2022-01-31 14:53:00,663 - octoprint.plugins.softwareupdate - INFO - Starting update of pip to 22.0.2...
2022-01-31 14:53:27,489 - octoprint.plugins.softwareupdate - INFO - Update of pip to 22.0.2 successful!
2022-01-31 14:53:27,692 - octoprint.plugins.softwareupdate - INFO - Starting update of DisplayLayerProgress to 1.28.0...
2022-01-31 14:53:28,368 - octoprint.plugins.tracking - INFO - Sent tracking event update_successful, payload: {'target': 'pip', 'from_version': '21.3.1', 'to_version': '22.0.2'}
2022-01-31 14:53:41,423 - octoprint.plugins.softwareupdate - ERROR - Update of DisplayLayerProgress can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 2090, in _perform_update
    target, populated_check, target_version, log_cb=self._log, online=online
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 115, in perform_update
    returncode, stdout, stderr = pip_caller.execute(*pip_args, **pip_kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/pip.py", line 208, in execute
    return self.call(command, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 244, in call
    all_stdout += process_stdout(p.stdout.readlines(timeout=0.5))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 236, in process_stdout
    return process_lines(lines, self._log_stdout)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 230, in process_lines
    *map(lambda x: to_unicode(x, errors="replace"), lines)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/pip.py", line 498, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/pip.py", line 517, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode characters in position 5-44: ordinal not in range(256)
2022-01-31 14:53:41,512 - octoprint.plugins.softwareupdate - INFO - Saved update log to disk
2022-01-31 14:53:42,301 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'target': 'DisplayLayerProgress', 'from_version': '1.27.2', 'to_version': '1.28.0'}
2022-01-31 15:00:48,516 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2022-01-31 15:01:21,110 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 5419, 'printer_state': 'OPERATIONAL'}
```

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Not applicable

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

0.17.0

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

Chrome latest, Win 7-64

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220131150911.zip](https://github.com/OctoPrint/OctoPrint/files/7973895/octoprint-systeminfo-20220131150911.zip)

foosel, 2022-02-01:
I don't see a failure in updating pip here, in fact I see a success for that. However, updating DisplayLayerProgress is failing. I've corrected the misleading title.

The encoding error being triggered is a new one that I've never seen before, and I'm wondering if by chance that might be caused by the pip plugin. If you would be so kind as to downgrade pip again to version 21.3.1 to rule this out as the culprit here, that would be very helpful. If the issue persists then, I might need to get some help from @OllisGit got get to the bottom of why this happens all of a sudden.

jcwren, 2022-02-01:
You're correct, I misread that and it's indeed the plug-in that's failing. Sorry about that.  I'm not a Python person, so would you mind giving me the command to downgrade pip? 

I'm assuming I'll have to issue the command `/home/pi/oprint/bin/python -m pip --disable-pip-version-check install https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/latest/download/master.zip --no-cache-dir` from the shell. Do I need to be in a particular directory when doing this?

jcwren, 2022-02-01:
Well, this is odd. I had tried the update one more time this morning and it failed again (as I suspected it would). I shut down the Pi to move where the printer is located, and when I brought it back up, it's no longer saying any updates are available, and DisplayLayerProgress is at 1.28.0. It appears it healed itself by simply rebooting.

I don't know if its related, but I tried the update a couple times, and each time it updated pip to 22.0.2, even though it had succeeded the previous times. 

Unless you'd like me to pursue this further, we can close this.

foosel, 2022-02-04:
After some further investigation by @cp2004 (thanks!) this seems to actually be rooted in an issue between latest pip's standard output and OctoPrint. So there is in fact a problem to solve here.

kantlivelong, 2022-02-04:
Perhaps `-no-color --progress-bar ascii` might do it?

kantlivelong, 2022-02-04:
https://github.com/pypa/pip/issues/10461

foosel, 2022-02-07:
@donnels Which version of pip?

donnels, 2022-02-07:
2022-02-07 14:28:04,845 - octoprint.plugins.softwareupdate - INFO - Starting update of pip to 22.0.3...
2022-02-07 14:28:14,598 - octoprint.plugins.softwareupdate - INFO - Update of pip to 22.0.3 successful!
2022-02-07 14:28:14,649 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.7.3...

foosel, 2022-02-07:
Can you please SSH into the machine, run this manually:

    ~/oprint/bin/python -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint/archive/1.7.3.zip --no-cache-dir

And then copy the *exact output* back here?

donnels, 2022-02-07:
as which user would you like it run?

foosel, 2022-02-07:
Regular `pi` user please.

donnels, 2022-02-07:
pi@octopi-ender:~ $ ~/oprint/bin/python -m pip --disable-pip-version-check install https://github.com/OctoPrint/OctoPrint/archive/1.7.3.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/OctoPrint/OctoPrint/archive/1.7.3.zip
  Downloading https://github.com/OctoPrint/OctoPrint/archive/1.7.3.zip (6.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.5/6.5 MB 1.1 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2021.2.23)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2021.10.11)
Requirement already satisfied: OctoPrint-PiSupport>=2021.10.28 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2021.10.28)
Requirement already satisfied: markupsafe<2.0,>=1.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (3.1.1)
Requirement already satisfied: wrapt<1.13,>=1.12.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.12.1)
Requirement already satisfied: flask<2,>=1.1.4 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.1.4)
Requirement already satisfied: Jinja2<3,>=2.11.3 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2.11.3)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.5.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.0.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.0.1)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.1.0)
Requirement already satisfied: cachelib<0.2,>=0.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.1.1)
Requirement already satisfied: PyYAML<6,>=5.4.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (5.4.1)
Requirement already satisfied: pyserial<4,>=3.4 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (3.5)
Requirement already satisfied: netaddr<0.9,>=0.8 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.8.0)
Requirement already satisfied: watchdog==0.10.4 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.10.4)
Requirement already satisfied: sarge==0.1.6 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.1.6)
Requirement already satisfied: netifaces<1,>=0.11 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.11.0)
Requirement already satisfied: pylru<2,>=1.2 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.2.0)
Requirement already satisfied: pkginfo<2,>=1.7.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.7.1)
Requirement already satisfied: requests<3,>=2.26.0 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2.26.0)
Requirement already satisfied: semantic_version<3,>=2.8.5 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2.8.5)
Requirement already satisfied: psutil<6,>=5.8 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (5.8.0)
Requirement already satisfied: Click<8,>=7.1.2 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (7.1.2)
Requirement already satisfied: future<1,>=0.18.2 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.18.2)
Requirement already satisfied: websocket-client<1,>=0.59 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.59.0)
Requirement already satisfied: emoji<2,>=1.4.2 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.6.1)
Requirement already satisfied: sentry-sdk<2,>=1.3.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.4.3)
Requirement already satisfied: filetype<2,>=1.0.7 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.0.7)
Requirement already satisfied: zipstream-new<1.2,>=1.1.8 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.1.8)
Requirement already satisfied: blinker<2,>=1.4 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (1.4)
Requirement already satisfied: feedparser<7,>=6.0.8 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (6.0.8)
Requirement already satisfied: tornado<7,>=6 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (6.1)
Requirement already satisfied: zeroconf<0.34,>=0.33 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.33.4)
Requirement already satisfied: immutabledict<3,>=2.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2.2.1)
Requirement already satisfied: pathvalidate<3,>=2.4.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (2.5.0)
Requirement already satisfied: colorlog<6,>=5.0.1 in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (5.0.1)
Requirement already satisfied: unidecode in ./oprint/lib/python3.7/site-packages (from OctoPrint==1.7.3) (0.4.21)
[... cut ...]
Requirement already satisfied: charset-normalizer~=2.0.0 in ./oprint/lib/python3.7/site-packages (from requests<3,>=2.26.0->OctoPrint==1.7.3) (2.0.7)
Requirement already satisfied: certifi>=2017.4.17 in ./oprint/lib/python3.7/site-packages (from requests<3,>=2.26.0->OctoPrint==1.7.3) (2020.12.5)
Requirement already satisfied: six in ./oprint/lib/python3.7/site-packages (from websocket-client<1,>=0.59->OctoPrint==1.7.3) (1.15.0)
Requirement already satisfied: ifaddr>=0.1.7 in ./oprint/lib/python3.7/site-packages (from zeroconf<0.34,>=0.33->OctoPrint==1.7.3) (0.1.7)

cp2004, 2022-02-07:
It's the same issue as this one, because it happened after you updated to Pip 22.

foosel, 2022-02-07:
Thanks, it's indeed this line:

```
━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.5/6.5 MB 1.1 MB/s eta 0:00:00
```

I'm still unable to reproduce this myself with upgraded pip, but at least I can now verify that my fix indeed works. THanks.

You should be able to get things going in the meantime again by *downgrading* `pip`, as described in https://github.com/OctoPrint/OctoPrint/issues/4392#issuecomment-1026944365

donnels, 2022-02-07:
Many thanks. If you need anything from this install feel free. I'll update/downgrade later.

foosel, 2022-02-07:
@donnels all good, thank you!

closed 2022-05-17 by foosel

rewolff, 2022-05-29:
I've been encountering this issue for a while now. Ending in "can't upgrade, check the log" and then tricky-to-parse error mesages. Last time I left it at that, but this time I searched for a key term, I think and found this. 

I don't know how "failed" the upgrade is when it ends like that, but following the suggestion above: "it suddenly worked after rebooting" I tried that. Now the system comes up and claims to be 1.8.1. I think I was upgrading to 1.8.1 when I was told the upgrade failed.

commit 36d6768f (2022-02-07): 🐛 Fix clean_ansi's unicode compatibility

Closes #4392