OctoPrint/OctoPrint #4395: Octoprint connects to "Nothing"

heregoesnothing3d, 2022-01-31:
### The problem

1. I boot up octoprint
2. Immediately (even if printer isn't plugged into pi) it says it is connected, but I don't know what to
3. When i hit disconnect, it takes ~10 minutes to disconnect, and then when I plug in my printer and hit connect, it connects fine

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

Linux on Pi 3B+

### Printer model & used firmware incl. version

Ender 3 V2 (Modded) with Marlin 

### Browser and version of browser, operating system running browser

Marlin 2.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220131180537.zip](https://github.com/OctoPrint/OctoPrint/files/7974929/octoprint-systeminfo-20220131180537.zip)
[image]

heregoesnothing3d, 2022-03-02:
FOLLOW UP: I thought I had this worked out, but no. Basically, it works fine on my other computer, so it is something with the connection on my laptop (the one with the original issue).

heregoesnothing3d, 2022-03-02:
> There's something wrong with the connection to the websocket here. In the log file, it is filled with messages going from 'new connection' -> 'logging in' -> 'connection closed' and around again, twice a second. So for some reason the connection is constantly being dropped, so it probably never gets to the fully 'connected' state where everything would be 'alive'.
> 
> Do you see the same behaviour on other devices, or on a different browser? Just to try and isolate the problem if possible. I can't see a log file from the safe mode test in there, was the system info bundle downloaded before you tried that? It looks like it's the same file from the other tickets created, so maybe before the question was asked in the template, which is understandable.

I tried other browser, but that doesn't work. It DOES work on my desktop, but I do not want to have to use that every time I want to start/stop a print haha (laptop is easier for me).

foosel, 2022-03-02:
Please start in safe mode, reproduce, provide a system info bundle. 

Also try another browser and a fresh browser profile on the laptop. Ideally also disable all browser add-ons to rule out any issues caused by them.

It certainly does sound more like an issue with your laptop, OctoPrint requires a stable connection to the server, and if the socket constantly closes it won't be able to function properly.

heregoesnothing3d, 2022-03-02:
> Please start in safe mode, reproduce, provide a system info bundle.
>
> Also try another browser and a fresh browser profile on the laptop. Ideally also disable all browser add-ons to rule out any [...]

Safe mode doesn't work. Tried using another browser and a new profile, but still same issue. I just found [this](https://community.octoprint.org/t/the-web-interface-seems-stuck-not-responsive-broken/5675) , but none of the solutions seem to make a difference.

jneilliii, 2022-03-02:
Have you tried accessing it on a broken device in incognito mode yet? You mention you can access from one machine, so starting OP in safe mode and providing the requested system bundle after attempting to access from the broken device would potentially help in diagnosing the problem.

heregoesnothing3d, 2022-03-02:
No luck. Same behavior in incog than with normal browser

foosel, 2022-03-03:
We *still* would like to have a system info bundle from a safe mode run with a working and non working browser.

foosel, 2022-03-03:
And since I just saw that you also cross posted this on Reddit, I'll close this ticket now. As you already noted yourself, all things point to this being some weird issue with your Laptop, and support requests like this should go to [the heavily frequented community forums](https://community.octoprint.org) anyhow and not contribute to the work load of the select few that monitor this bug tracker.

closed 2022-03-03 by foosel