OctoPrint/OctoPrint #4396: Standard output is unreadable on console

zaitcev, 2022-02-01:
### The problem

I didn't do a git pull for a few months and somehow the console output became colorized so much that it's basically unreadable. This was okay before, at least in October 2021.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

lI

### Printer model & used firmware incl. version

Linux, Fedora 35

### Browser and version of browser, operating system running browser

Firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220131185951.zip](https://github.com/OctoPrint/OctoPrint/files/7974982/octoprint-systeminfo-20220131185951.zip)
[image]

cp2004, 2022-02-01:
The log output was colourised starting with 1.7.x. Took a peek at the code, it is setting 'INFO' to be white. On a white background, that definitely doesn't work so good...

Maybe a configurable flag could be used (`--no-colour` or something) to disable it, or I will test what happens if the 'INFO' colour is unset, so it would go back to being the 'default'.

zaitcev, 2022-02-01:
I would think that killing colors altogether may be throwing out the baby with the bathwater. However, it is still preferable for my circumstances.

cp2004, 2022-02-05:
Addressed by #4402

zaitcev, 2022-02-09:
I'm sorry if I'm missing something, but it appears that the fixes were merged into branch "maintenance", which I'm using branch "master", so they are not available to me. Are the commits in danger of being forgotten in the maintenance branch?

cp2004, 2022-02-09:
master is the current stable release, aka OctoPrint 1.7.3. When 1.8.0 is released, the work that is currently ongoing to be 1.8.0 that is happening on the maintenance branch will be merged into the master branch as the new stable release.

foosel, 2022-02-09:
Also refer to the branching model described in the docs here: https://docs.octoprint.org/en/master/development/branches.html and linked to from the contribution guidelines.

zaitcev, 2022-02-09:
Thanks for the explanation and the reference. I'll close the issue once I'm able to test.

commit e9c33e1f (2022-02-09): ✨  Add `--no-color` option to disable colored output on console (#4402)

* :sparkles: Add `--no-color` option to disable colored output on console

Closes #4396

Also removed the setting of INFO logs to white, instead it will be left as default. If there is a light-colored background (such as white) then it becomes unreadable. Can then just use whatever user has configured for their default and not worry about contrast issues.

* 🎨 Remove redundant bool wrapper

* ⬆️ Update to colorlog 6

That way we get support for NO_COLOR as well:

https://no-color.org/

* 💬 Add NO_COLOR to --no-color help
closed 2022-05-17 by foosel