OctoPrint/OctoPrint #4397: [Request] Can the GCODEVIEWER be sensitive to G1 with S parameter

felinh, 2022-02-01:
### Is your feature request related to a problem? Please describe.

Hi , thank so much for this wonderful tool. I use it every day.
I have adapted an old 3D printer to laser and I am using Gcode with S parameter instead of "E"
like this :G1 X19.850 Y34.350 S50

### Describe the solution you'd like

I wish I can use the GCODE viewer to see the LASER paths but it seems it required G1 with 'E' to get visualisation.

### Describe alternatives you've considered

I wonder if I overload my code with fake G1 + Exxx it would work....

foosel, 2022-02-01:
OctoPrint first and foremost targets 3D printers, and they don't know anything about `S` parameters on `G1` commands. So this is not a change I'd like to see nor maintain in OctoPrint for now.

However, you might be able to still achieve your goal with a custom plugin that rewrites the files on upload to have `E` coordinates where there are `S` coordinates, and maybe also strip them away again on transmission to your machine (or just leave them in if the firmware doesn't care about an `E` there).

closed 2022-02-01 by foosel

felinh, 2022-02-01:
OK. workaround works anyway.
generating 'fake' extrusion command with increasing E value and 'S' works for  both for MARLIN based lasers and native octoprint GCODE viewer  
Exemple:
G1 X4.800 Y5.050 F600.0 E0.31 S50
G1 X4.800 Y4.800 E0.32 S50
G1 X5.050 Y4.800 E0.33 S50

awful but efficient :-)