OctoPrint/OctoPrint #4398: Prusa MK3 "enqueuing to front M600"

midnightsmith89, 2022-02-02:
### The problem

almost any print sent with octoprint (1.7.2) results in the same issue of after a bed probe the extruder parks and begins an m600 filament change. It "unloads" then loads a random position on the MMU, then unloads that and loads the proper position and asks if it was loaded correctly. It then re probes the bed and begins printing.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.2

### Operating system running OctoPrint

OctoPi 0.18

### Printer model & used firmware incl. version

MK3s with MMU2S latest versions

### Browser and version of browser, operating system running browser

Chrome, windows 10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220113043920.zip](https://github.com/OctoPrint/OctoPrint/files/7982770/octoprint-systeminfo-20220113043920.zip)
[serial (1).log](https://github.com/OctoPrint/OctoPrint/files/7982771/serial.1.log)
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint/files/7982772/octoprint.1.log)

please see the issue here https://community.octoprint.org/t/prusa-mk3s-w-mmu-enqueing-to-front-m600/40734/3

foosel, 2022-02-02:
Sorry, but you'll have to take this up with the firmware devs, because this is clearly not an OctoPrint issue but rather the firmware injecting commands on its own after triggering resend requests for lines it already happily acknowledged prior. Something's amiss there.

See this annotated excerpt from your system info bundle's serial.log:

```
2022-01-13 04:32:32,939 - Changing monitoring state from "Operational" to "Starting"
2022-01-13 04:32:32,987 - Send: N0 M110 N0*125
2022-01-13 04:32:32,994 - Recv: ok // N0
2022-01-13 04:32:32,995 - Changing monitoring state from "Starting" to "Printing"
2022-01-13 04:32:32,996 - Send: N1 M73 P0 R15*32
2022-01-13 04:32:33,010 - Recv: NORMAL MODE: Percent done: 0; print time remaining in mins: 15; Change in mins: -1
2022-01-13 04:32:33,015 - Recv: SILENT MODE: Percent done: -1; print time remaining in mins: -1; Change in mins: -1
2022-01-13 04:32:33,016 - Recv: ok // N1
2022-01-13 04:32:33,017 - Send: N2 M73 Q0 S15*35
2022-01-13 04:32:33,028 - Recv: NORMAL MODE: Percent done: 0; print time remaining in mins: 15; Change in mins: -1
2022-01-13 04:32:33,036 - Recv: SILENT MODE: Percent done: 0; print time remaining in mins: 15; Change in mins: -1
2022-01-13 04:32:33,036 - Recv: ok // N2
2022-01-13 04:32:33,037 - Send: N3 M201 X1000 Y1000 Z200 E8000*7
2022-01-13 04:32:33,044 - Recv: ok // N3
2022-01-13 04:32:33,045 - Send: N4 M203 X200 Y200 Z12 E120*8
2022-01-13 04:32:33,052 - Recv: ok // N4
2022-01-13 04:32:33,053 - Send: N5 M204 P1250 R1250 T1250*80
2022-01-13 04:32:33,060 - Recv: ok // N5
2022-01-13 04:32:33,061 - Send: N6 M205 X8.00 Y8.00 Z0.40 E4.50*57
2022-01-13 04:32:33,068 - Recv: ok // N6
2022-01-13 04:32:33,069 - Send: N7 M205 S0 T0*36
2022-01-13 04:32:33,072 - Recv: ok // N7
2022-01-13 04:32:33,073 - Send: N8 M107*45
2022-01-13 04:32:33,077 - Recv: ok // N8
2022-01-13 04:32:33,077 - Send: N9 M862.3 P "MK3SMMU2S"*57
2022-01-13 04:32:33,085 - Recv: ok // N9
2022-01-13 04:32:33,086 - Send: N10 M862.1 P0.6*89
2022-01-13 04:32:33,089 - Recv: ok // N10
2022-01-13 04:32:33,090 - Send: N11 M115 U3.10.0*97
2022-01-13 04:32:33,093 - Recv: ok // N11
2022-01-13 04:32:33,094 - Send: N12 G90*35
2022-01-13 04:32:33,097 - Recv: ok // N12
2022-01-13 04:32:33,098 - Send: N13 M83*42
2022-01-13 04:32:33,101 - Recv: ok // N13
2022-01-13 04:32:33,103 - Send: N14 M104 S210*83
2022-01-13 04:32:33,109 - Recv: ok // N14
2022-01-13 04:32:33,110 - Send: N15 M140 S60*103
2022-01-13 04:32:33,114 - Recv: ok // N15
2022-01-13 04:32:33,114 - Send: N16 M190 S60*105
2022-01-13 04:32:33,122 - Recv: LCD status changed
2022-01-13 04:32:33,122 - Recv: LCD status changed
2022-01-13 04:32:33,122 - Recv: ok // N16
2022-01-13 04:32:33,123 - Send: N17 M109 S210*93
2022-01-13 04:32:33,130 - Recv: LCD status changed
2022-01-13 04:32:34,130 - Recv: T:211.8 E:0 W:?
[...]
2022-01-13 04:32:39,144 - Recv: T:210.6 E:0 W:0
2022-01-13 04:32:39,635 - Recv: LCD status changed
2022-01-13 04:32:39,636 - Recv: ok // N17
2022-01-13 04:32:39,638 - Send: N18 G28 W*93
2022-01-13 04:32:39,762 - Recv: tmc2130_home_enter(axes_mask=0x01)
2022-01-13 04:32:40,237 - Recv: T:209.9 /210.0 B:60.6 /60.0 T0:209.9 /210.0 @:41 B@:15 P:36.2 A:41.7
2022-01-13 04:32:41,642 - Recv: echo:busy: processing
2022-01-13 04:32:41,831 - Recv:   0 step=45 mscnt= 734
[...]
2022-01-13 04:32:47,643 - Recv: echo:busy: processing
2022-01-13 04:32:48,261 - Recv: T:209.7 /210.0 B:60.4 /60.0 T0:209.7 /210.0 @:32 B@:40 P:36.2 A:41.7
2022-01-13 04:32:49,388 - Recv: ok // N18
2022-01-13 04:32:49,390 - Send: N19 G80*41
2022-01-13 04:32:50,268 - Recv: T:209.7 /210.0 B:60.3 /60.0 T0:209.7 /210.0 @:31 B@:51 P:36.2 A:41.7
2022-01-13 04:32:51,402 - Recv: echo:busy: processing
[...]
2022-01-13 04:33:43,403 - Recv: echo:busy: processing
2022-01-13 04:33:44,415 - Recv: T:209.7 /210.0 B:60.9 /60.0 T0:209.7 /210.0 @:32 B@:0 P:37.0 A:41.4
2022-01-13 04:33:44,424 - Recv: LCD status changed
[... 217 lines cut ...]
2022-01-13 04:38:27,120 - Recv: NORMAL MODE: Percent done: 2; print time remaining in mins: 15; Change in mins: -1
2022-01-13 04:38:27,128 - Recv: SILENT MODE: Percent done: 1; print time remaining in mins: 15; Change in mins: -1
2022-01-13 04:38:27,128 - Recv: ok // N40
2022-01-13 04:38:27,129 - Send: N41 M84*42
2022-01-13 04:38:27,132 - Recv: Unknown M code: $40 M108
2022-01-13 04:38:27,132 - Recv: ok // N41, even though it apparently it arrived scrambled or was scrambled on reception, buffer issues from earlier?
2022-01-13 04:38:27,134 - Send: N42 M104 T0 S0*23
2022-01-13 04:38:27,201 - Recv: T:211.0 /210.0 B:60.0 /60.0 T0:211.0 /210.0 @:27 B@:29 P:36.9 A:40.7
2022-01-13 04:38:29,135 - Recv: echo:busy: processing
[...]
2022-01-13 04:38:39,134 - Recv: echo:busy: processing
2022-01-13 04:38:39,236 - Recv: T:211.4 /210.0 B:60.1 /60.0 T0:211.4 /210.0 @:31 B@:8 P:37.1 A:41.0
2022-01-13 04:38:40,138 - Recv: ok // N42
2022-01-13 04:38:40,140 - Send: N43 M140 S0*82
2022-01-13 04:38:40,147 - Recv: ok // N43
2022-01-13 04:38:40,150 - Send: N44 M106 S0*87
2022-01-13 04:38:40,151 - Recv: ok // N44
2022-01-13 04:38:40,153 - Changing monitoring state from "Cancelling" to "Operational"
```

Annotations by me via `//`. OctoPrint happily sends stuff over line by line, gets an acknowledgment for each line up until 38, when the firmware suddenly decides it didn't like the already acknowledged line 33 after all and then it keeps requesting that for a while, all the while happily enqueuing `M600` internally. You cancel the print during this, the firmware doesn't even yiel [... cut ...]
Firmware issue, there's nothing that OctoPrint could have done differently here that would make this magically work. Just to rule out any kind of issues due to EMI, check all your communication wiring (Pi <-> printer and printer <-> MMU) is well routed, NOT next to motors or anything like that that might cause interferences on the line, but that's about the only advice I can give you here.

There's not even a hint of a bug in OctoPrint here, so I'm closing this ticket and suggest to continue this on the forums if you have further questions, though as I said, this is most likely a firmware issue, either on the printer or the MMU side of things.

closed 2022-02-02 by foosel

sliverbaer, 2022-08-07:
Howdy,  I just ran into this issue.  (I have had it happen before, without using OctoPrint.)

`Recv: fsensor_checkpoint_print
Recv: ok
Recv: Resend: 648069
Recv: ok
Recv: LCD status changed
Recv: fsensor_update - M600
Send: N648069 G1 X82.471 Y116.079 E-.27367*125
Recv: echo:Enqueing to the front: "M600"
Recv: T:214.5 /215.0 B:59.9 /60.0 T0:214.5 /215.0 @:78 B@:38 P:0.0 A:40.8`

Filament is measuring at 1.65 mm and not 1.75 mm.  I believe this to be the issue when this happens.  Triggers the filament sensor of the MK3S+ to issue a filament change.  

Just wanted to post this so that others that search for it should measure or change their filament.