OctoPrint/OctoPrint #4401: webcam access not restricted

rico001, 2022-02-05:
### The problem

Hello,
It seems webcam access is not restricted.
Problem: /webcam/..-URL's are accessable without user login. I'm not sure that users are aware of this.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Anycubic

### Browser and version of browser, operating system running browser

all common browsers

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

fieldOfView, 2022-02-05:
The `/webcam' handling, and `mjpgstreamer` are not features of OctoPrint. They come with OctoPi, so your issue is with that repository. A quick search resulted in this feature request: https://github.com/guysoft/OctoPi/issues/292

closed 2022-02-05 by fieldOfView

foosel, 2022-02-05:
For the record, this is also covered in the FAQ

https://community.octoprint.org/docs?topic=233

and OctoPrint also tells the user that exposing the instance in the public internet or other hostile networks is a really bad idea at multiple places, this being one of the reasons.

cp2004, 2022-02-05:
And there is the warning at the top of the page here:
[image]

rico001, 2022-02-05:
I added my cam with plug an play yesterday. I did not know this cam tab in settings. thx guys.