OctoPrint/OctoPrint #4406: Communication with printer freeze - Anycubic i3 with OctoPrint

ksergeother, 2022-02-07:
### The problem

I have install my octopi with raspberry pi 3b. The installation was successfull, but when i try to connect, i am getting one error.
For the story i have installed the plugins fix bcd. I am have the same configuration with octoprint  on widows ans everything works, but not with raspbrry. when i check the terminal i see this Repl: M110 N0 -> M110
Send: N0 M110*35 and the connection with octoprint is not working. i tried single baudrate without success.
I need help.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

Octopi 0.18.0

### Printer model & used firmware incl. version

Anycubic i3 Mega V1.4.1RC1/P2

### Browser and version of browser, operating system running browser

Chrome Version 97.0.4692.99

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

he printers currently known to ship with this broken CBD firmware. The plugin is not helping with octopi.
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 256000
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0x6f1cec90, open=True>(port='/dev/ttyUSB0', baudrate=256000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Repl: M110 N0 -> M110
Send: N0 M110*35
No answer from the printer within the connection timeout, trying another hello
Repl: M110 N0 -> M110
Send: N0 M110*35
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor

foosel, 2022-02-07:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2022-02-07 by foosel

ksergeother, 2022-02-07:
Thank you, i will update my request

ksergeother, 2022-02-20:
i was able to fix the issue following the link below. Hurraa!
https://github.com/OctoPrint/OctoPrint/issues/2309