OctoPrint/OctoPrint #4409: DietPi 8 + OctoPrint + Klipper + Creality Ender 3 Pro | Serial comms problem

eyduh, 2022-02-08:
### The problem

Not sure if this is an octoprint issue or a klipper issue so cross-posting here. The original issue was reported here:
[Bug report on Klipper3D github](https://github.com/Klipper3d/klipper/issues/5246#issue-1127289468) There's a lot of info regarding the setup and users/permissions in that issue report.

I originally setup my Creality Ender 3 Pro, 32-bit STM mainboard (v4.2.2), using the Creality version of marlin from their website and octoprint through the DietPi scripts `dietpi-software` and I could connect to the printer. There was the problem of incorrect temperatures reported (no temps) but that was fixed with the right plugins.

After this I was trying to setup BLTouch and had some trouble getting it to work and in troubleshooting that I stumbled across the custom firmware Klipper and decided to start over with that as it seemed better in many ways. There were also some people on that forum and here on github who seemed to have gotten BLTouch to work so yay!

After some initial troubleshooting I managed to get klipper talking to the printer with the new klipper firmware. The LCD went from blank to the klipper interface. So, I moved on to the part where I needed to add /tmp/printer to octoprint and here I ran into some issues. It seems that `comm.py` can't talk to the printer.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

DietPi 8 (Debian Bullseye)

### Printer model & used firmware incl. version

Creality Ender 3 Pro, mainboard 4.2.2, klipper.bin firmware

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Systeminfo bundle
[octoprint-systeminfo-20220208175155.zip](https://github.com/OctoPrint/OctoPrint/files/8025842/octoprint-systeminfo-20220208175155.zip)

```
$ octoprint systeminfo
connectivity.connection_check: 9.9.9.9:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 1500.0
env.hardware.ram: 933044224
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 22.0.3
env.python.version: 3.9.2
env.python.virtualenv: False
octoprint.version: 1.7.3
systeminfo.generator: cli
```

excerpt from the web-logs whether trying to connect to /dev/ttyUSB0 (which is a symlink to the actual path by-id) or /tmp/printer (which is a link to whatever is in the printer.cfg file loaded by klipper).
```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyUSB0@250000, /dev/ttyUSB0@230400, /dev/ttyUSB0@57600, /dev/ttyUSB0@38400, /dev/ttyUSB0@19200, /dev/ttyUSB0@9600
Trying port /dev/ttyUSB0, baudrate 115200
Connecting to port /dev/ttyUSB0, baudrate 115200
Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 115200 from hook default: SerialException: '[Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3775
Could not open port /dev/ttyUSB0, baudrate 115200, skipping
[... 150 lines cut ...]
Feb 08 17:55:57 dietprints0 octoprint[8750]: 2022-02-08 17:55:57,761 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port /dev/ttyUSB0, baudrate 9600 from hook default: SerialException: '[Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable' @ comm.py:_open_serial:3775
Feb 08 17:55:57 dietprints0 octoprint[8750]: Traceback (most recent call last):
Feb 08 17:55:57 dietprints0 octoprint[8750]:   File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.9/site-packages/serial/serialposix.py", line 385, in _reconfigure_port
Feb 08 17:55:57 dietprints0 octoprint[8750]:     fcntl.flock(self.fd, fcntl.LOCK_EX | fcntl.LOCK_NB)
Feb 08 17:55:57 dietprints0 octoprint[8750]: BlockingIOError: [Errno 11] Resource temporarily unavailable
Feb 08 17:55:57 dietprints0 octoprint[8750]: During handling of the above exception, another exception occurred:
Feb 08 17:55:57 dietprints0 octoprint[8750]: Traceback (most recent call last):
Feb 08 17:55:57 dietprints0 octoprint[8750]:   File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.9/site-packages/octoprint/util/comm.py", line 3775, in _open_serial
Feb 08 17:55:57 dietprints0 octoprint[8750]:     serial_obj = factory(
Feb 08 17:55:57 dietprints0 octoprint[8750]:   File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.9/site-packages/octoprint/util/comm.py", line 3740, in default
Feb 08 17:55:57 dietprints0 octoprint[8750]:     serial_obj.open()
Feb 08 17:55:57 dietprints0 octoprint[8750]:   File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.9/site-packages/serial/serialposix.py", line 332, in open
Feb 08 17:55:57 dietprints0 octoprint[8750]:     self._reconfigure_port(force_update=True)
Feb 08 17:55:57 dietprints0 octoprint[8750]:   File "/mnt/dietpi_userdata/octoprint/.local/lib/python3.9/site-packages/serial/serialposix.py", line 387, in _reconfigure_port
Feb 08 17:55:57 dietprints0 octoprint[8750]:     raise SerialException(msg.errno, "Could not exclusively lock port {}: {}".format(self._port, msg))
Feb 08 17:55:57 dietprints0 octoprint[8750]: serial.serialutil.SerialException: [Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable
Feb 08 17:55:57 dietprints0 octoprint[8750]: 2022-02-08 17:55:57,762 - octoprint.util.comm - INFO - Serial detection: Could not open port /dev/ttyUSB0, baudrate 9600, skipping
Feb 08 17:55:57 dietprints0 octoprint[8750]: 2022-02-08 17:55:57,762 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error"
Feb 08 17:55:57 dietprints0 octoprint[8750]: 2022-02-08 17:55:57,763 - octoprint.util.comm - INFO - Changing monitoring state from "Error" to "Offline after error"
Feb 08 17:55:57 dietprints0 octoprint[8750]: 2022-02-08 17:55:57,853 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
Feb 08 17:55:57 dietprints0 octoprint[8750]: 2022-02-08 17:55:57,976 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_autodetect, payload: {'commerror_text': 'No more candidates to test, and no working port/baudrate combination detected.'}

``` 

Since klipper can find and connect to the printer I suspect this is an octoprint issue with the klipper firmware?

cp2004, 2022-02-08:
> So, I moved on to the part where I needed to add /tmp/printer to octoprint and here I ran into some issues. 

You still need to do this part. OctoPrint needs to connect to this to communicate with Klipper. It doesn't connect to the USB port, which is what all your logs are showing it is trying to do. There's no bug here, this would be best sorted on the forums or discord server.

cp2004, 2022-02-08:
And I clicked through to the Klipper ticket, where there was the useful info. If Klipper and OctoPrint are not running as the same user, then it would be expected that OctoPrint does not have permission to open Klipper's virtual serial port.

The issue is with neither OctoPrint nor Klipper, it is the way they have been setup.

eyduh, 2022-02-09:
> If Klipper and OctoPrint are not running as the same user, then it would be expected that OctoPrint does not have permission to open Klipper's virtual serial port.

If the virtual serial port has the proper Group permissions and the two users are in each other's groups it -should- still work no? 

As you can see all the symbolic links have 777 as expected under GNU/Linux, the files they point to however all have 660 which still allows to user & group read/write access.

```
$ id octoprint
uid=997(octoprint) gid=995(octoprint) groups=995(octoprint),5(tty),20(dialout),44(video),1000(dietpi)
$ id dietpi
uid=1000(dietpi) gid=1000(dietpi) groups=1000(dietpi),5(tty),20(dialout),44(video),995(octoprint)

$ groups octoprint
octoprint : octoprint tty dialout video dietpi
$ groups dietpi 
dietpi : dietpi tty dialout video octoprint
```

Both users are in each other's groups and tty and dialout

All the different links/files are permissive enough (660 or user & group rw permissions)
```
$ stat /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
  File: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0
  Size: 13              Blocks: 0          IO Block: 4096   symbolic link
Access: (0777/lrwxrwxrwx)  Uid: (    0/    root)   Gid: (    0/    root)

$ stat /dev/ttyUSB0
  File: /dev/ttyUSB0
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Access: (0660/crw-rw----)  Uid: (    0/    root)   Gid: (   20/ dialout)

$ stat /tmp/printer 
  File: /tmp/printer -> /dev/pts/0
  Size: 10              Blocks: 0          IO Block: 4096   symbolic link
Device: 1ch/28d Inode: 6360        Links: 1
Access: (0777/lrwxrwxrwx)  Uid: ( 1000/  dietpi)   Gid: ( 1000/  dietpi)

$ stat /dev/pts/0
  File: /dev/pts/0
  Size: 0               Blocks: 0          IO Block: 1024   character special file
Device: 14h/20d Inode: 3           Links: 1     Device type: 88,0
Access: (0660/crw-rw----)  Uid: ( 1000/  dietpi)   Gid: (    5/     tty)

```

So, it _shouldn't_ matter if the processes run as different users no? 

octoprint is installed the way it is described for raspbian on the forum, klipper as they describe on their how to, with the addition of adding to each other's groups and tty/dialout.

> You need to connect to /tmp/printer at **all** times.
I suppose you mean at all times in octoprint right? In the Klipper configs from the git the serial points to the path by-id tho so pointing that to /tmp/printer would set up a beautiful loop.

> There's no bug here, this would be best sorted on the forums or discord server.
Every project has different policies re what goes where, since it worked with one firmware but not the other I put it here. Discord means future people with similar problems can't find any solutions and have to do the same work all over again. I had a look around the klipper forum before posting on that github and cross-posting here was easy.

---

> Does the issue persist if you uncheck "Request exclusive access to the serial port" under Settings > Serial Connection > General > Connection?
Problem persists, both under safe mode and normal.

foosel, 2022-02-09:
Charlie raised a good point here, I overlooked that you are apparently attempting to connect to the printer instead of Klipper when I suggested to try the mentioned settings change, and you should actually be connecting to `/tmp/printer`. The USB serial is no longer of interest to OctoPrint in a Klipper setup.

As far as I understand this, your `/dev/ttyUSB0` aka `/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0` is what *Klipper* connects to, no? Klipper then creates a virtual serial at `/tmp/printer` to which OctoPrint or any other host should connect to. OctoPrint can't talk to your printer. First of all Klipper is already talking to it (hence the exclusive access error), second of all even if it could it wouldn't be able to communicate with it since your printer now runs Klipper's firmware side of things and that's not a protocol OctoPrint understands (or should ever understand, as that would kinda defeat the whole point of the server portion of Klipper).

Add `/tmp/printer` to the list of additional ports if not already done, do whatever else additional setup the Klipper docs recommend to do in OctoPrint to make this work, try connecting to that, and only that, from OctoPrint. So select it and the correct baudrate manually from the connection drop down, do not attempt auto connection as that will try the USB stuff first, unless you blacklist that. 

If it doesn't work then, please get in touch on the forums. The issue tracker in this project is reserved for bug reports and feature requests, the forums (and Discord for people who prefer that) are for support and general discussion ([please also refer to CONTRIBUTING.md](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md)). Thank you.

closed 2022-02-09 by foosel

eyduh, 2022-02-09:
> Add `/tmp/printer` to the list of additional ports if not already done, do whatever else additional setup the Klipper docs recommend to do in OctoPrint to make this work, try connecting to that, and [...]

I've done this as stated in the summary her and in the klipper issue.

foosel, 2022-02-09:
I'm sorry, but then I misunderstood your summary:

> So, I moved on to the part where I needed to add /tmp/printer to octoprint and here I ran into some issues

This combined with repeated mentions of `/dev/ttyUSB0` and nothing of `/tmp/printer` in your actual connection attempts logs:

> Performing autodetection with 7 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyUSB0@250000, /dev/ttyUSB0@230400, /dev/ttyUSB0@57600, /dev/ttyUSB0@38400, /dev/ttyUSB0@19200, /dev/ttyUSB0@9600

looked like you had not in fact even tried connecting to `/tmp/printer`, or ran into an auto connection issue that I wanted to help you work around by forgoing the AUTO port and instead selecting the proper one as pointed out in my last comment. 

Glad to know it seems to work now though.