OctoPrint/OctoPrint #4410: [Maintenance] Some cached requests fail with `TypeError: can't compare offset-naive and offset-aware datetimes`

cp2004, 2022-02-08:
### The problem

Since upgrading all of the Flask ecosystem (I think, or maybe just the Py3 only stuff in general last week) I've been seeing this error occasionally popping up, leading to the requests failing:

```
2022-02-08 21:47:40,705 - octoprint - ERROR - Exception on /api/files [GET]
Traceback (most recent call last):
  File "C:\Users\charl\Development\venv310\lib\site-packages\flask\app.py", line 2073, in wsgi_app
    response = self.full_dispatch_request()
  File "C:\Users\charl\Development\venv310\lib\site-packages\flask\app.py", line 1518, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "C:\Users\charl\Development\venv310\lib\site-packages\flask\app.py", line 1516, in full_dispatch_request
    rv = self.dispatch_request()
  File "C:\Users\charl\Development\venv310\lib\site-packages\flask\app.py", line 1502, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "c:\users\charl\development\octoprint core\octoprint\src\octoprint\vendor\flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
  File "c:\users\charl\development\octoprint core\octoprint\src\octoprint\server\util\flask.py", line 1320, in decorated_function
    if condition(lm, etag) and not unless():
  File "c:\users\charl\development\octoprint core\octoprint\src\octoprint\server\util\flask.py", line 1296, in condition
    return check_lastmodified(lm) and check_etag(etag)
  File "c:\users\charl\development\octoprint core\octoprint\src\octoprint\server\util\flask.py", line 1383, in check_lastmodified
    and lastmodified <= flask.request.if_modified_since
TypeError: can't compare offset-naive and offset-aware datetimes
```

It's not just `/api/files`, but all of the cached functions seem to be failing at several points, working at others.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0.dev221+gf849315c1 - note that the log files will also show 1.7.3, this was actually a checkout of `devel` but for some reason it reported 1.7.3.

### Operating system running OctoPrint

Windows 11

### Browser and version of browser, operating system running browser

Microsoft Edge Version 98.0.1108.43 (Official build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220208220200.zip](https://github.com/OctoPrint/OctoPrint/files/8027794/octoprint-systeminfo-20220208220200.zip)

foosel, 2022-02-09:
The last modified data available on the request instance has been made timezone aware in Flask 2.0. All the local comparison datetime objects however aren't.

Now, I'm unsure how best to solve this. I can of course just convert everything in core and plugins under my control to call the caching decorator with a properly timezone aware datetime object. But what shall we do about third party plugins then?

I guess an error resilient fallback to timezone unaware comparison is needed here as well.

foosel, 2022-02-10:
I have not been able to reproduce this myself, but I *think* it should be fixed now with the above commit. Can you verify?

commit 5a5460d0 (2022-02-15): 🐛 Make sure lastmodified is timezone aware

Otherwise we'll run into an error comparing with
Flask's timezone aware headers.

Fixes #4410