OctoPrint/OctoPrint #4415: [BUG?] Server is offline locally. Fine on LAN. Print sometimes finishes, sometime frozes too.

bolsoncerrado, 2022-02-11:
### The problem

Extension of what I thought it was issue [2647](https://github.com/OctoPrint/OctoPrint/issues/2647), I started documenting weird behaviours here:
https://github.com/OctoPrint/OctoPrint/issues/2647#issuecomment-1033718970

HOWEVER today, I arrived at the farm and one of the Prusas was reporting SERVER OFFLINE locally, but I was able to access it thru the LAN from another computer. The print was going on.

[image]

[image]

My issue now is that I actually updated buck converters to 5A and this particular printer was NOT complaining of undervoltage, while one of its neighbours is (and completed print fine and still operative so far).

HENCE what I report is that for UNKNOWN reasons, I went from a fully working farm using Octopi 0.16/Python2 to a random hangs Octopi 0.18/Python 3 and Octoprint 1.7.3

I'm really puzzled now as the hardware is essentially the same, SDs were wiped and put a fresh image of Octopi 0.18, basically using any plugins other than stock, TouchUI and Excluderegion and I get this weird behaviour!

I ask for HELP!

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Live farm environment, haven't had the time to test in safe mode.

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

OctoPi 0.18

### Printer model & used firmware incl. version

Prusa MK3S+

### Browser and version of browser, operating system running browser

Chrome 98 on MacOS BigSur. Issues locally with TouchUI though

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Terminal shows normal printer temp "pings".

SSH to the Pi server is terribly slow.

[octoprint-systeminfo-20220211101012.zip](https://github.com/OctoPrint/OctoPrint/files/8047317/octoprint-systeminfo-20220211101012.zip)

[octoprint-logs (1).zip](https://github.com/OctoPrint/OctoPrint/files/8047324/octoprint-logs.1.zip)

Top shows this:

`top - 10:09:26 up 1 day,  1:08,  1 user,  load average: 10.70, 11.01, 9.15
Tasks: 147 total,   1 running, 146 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1.6 us,  5.6 sy,  0.0 ni,  1.4 id, 91.1 wa,  0.0 hi,  0.4 si,  0.0 st
MiB Mem :    871.7 total,     23.8 free,    740.5 used,    107.3 buff/cache
MiB Swap:    100.0 total,      0.0 free,    100.0 used.     19.5 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                              
   65 root      20   0       0      0      0 S   8.7   0.0  11:33.91 kswapd0                                                              
  434 pi        20   0  307500  54252   3152 D   5.2   6.1 205:57.38 octoprint                                                            
11114 root       0 -20       0      0      0 D   3.0   0.0   1:40.75 kworker/0:2H+kblockd                                                 
  852 pi        20   0  585712  17720   3996 D   1.4   2.0   8:01.07 chromium-browse                                                      
 1008 pi        20   0  298020  11628   3960 S   1.4   1.3   2:10.47 chromium-browse                                                      
 1083 pi        20   0 1067480 612304   1932 D   1.1  68.6   1006:06 chromium-browse                                                      
22078 pi        20   0   10400   1920   1256 R   1.1   0.2   0:00.14 top                                                                  
14362 root      20   0   52032  10392   2320 D   0.8   1.2   2:30.47 apt-get                                                              
   11 root      20   0       0      0      0 S   0.5   0.0   0:30.53 ksoftirqd/0                                                          
  125 root      20   0   40584   3504   3040 S   0.5   0.4   8:08.49 systemd-journal                                                      
  382 root      20   0   25512    940     44 S   0.5   0.1   2:28.15 rsyslogd                                                             
  506 root      20   0   10968    368    128 S   0.5   0.0   1:03.36 wpa_supplicant                                                       
21690 root       0 -20       0      0      0 I   0.5   0.0   0:02.15 kworker/0:1H-kblockd                                                 
21815 root      20   0       0      0      0 I   0.5   0.0   0:00.40 kworker/0:0-events                                                   
22106 root      20   0    3764   1640   1456 D   0.5   0.2   0:00.02 mjpg_streamer                                                        
    1 root      20   0   33812   1700    584 S   0.3   0.2  10:12.23 systemd                                                              
   12 root      20   0       0      0      0 I   0.3   0.0   1:15.51 rcu_sched                                                            
  707 haproxy   20   0   16080   2940   1848 S   0.3   0.3   1:01.20 haproxy                                                              
 1058 pi        20   0  270980   2748      0 S   0.3   0.3   0:02.04 chromium-browse                                                      
22077 root      20   0    7808   1272   1020 D   0.3   0.1   0:00.09 webcamd                     `

bolsoncerrado, 2022-02-11:
So whatever it is, it hogs the server until exthenuation and either kills the serial connection (sometimes!) or ends correctly.

foosel, 2022-02-11:
Please get in otuch on the community forums. The bug tracker is reserved for actual bugs and this looks more like an issue with your specific setup (we can always reopen this should it turn out to actually be a bug). There are WAY too many possible factors at play here given your dashboards, power setup and plugin masses. Also, re the latter, you *must* test in safe mode, otherwise it is not possible to determine if this is even an issue with vanilla OctoPrint or caused by a third party plugin.

closed 2022-02-11 by foosel

bolsoncerrado, 2022-02-11:
OOC why is 2647 even open still then?

foosel, 2022-02-11:
Because that's about the server crashing, and *only* the server crashing, your's is about a systemic issue where stuff is slow, disconnects, you can't even SSH in as mentioned in the other ticket, specifically in your farm setup (which I remember also was a bit special here and there from past tickets).

But yes, the issue tracker needs a cleanup indeed.

bolsoncerrado, 2022-02-11:
Well today is the only day the connection persisted, but eventually would have died perhaps, as much as the others.... I was finally able to SSH and grab a copy of the top output too...

And so far and once again, there's nothing special in my setup anymore as I learned the lesson, as you said, and minimized the amount of plugins (just those 2 and a few of yours), no webcams, no haproxy, no nothing else. My farm was working fine with the 0.16/Python 2 combo and now I get this after installing 0.18/Python3 FRESH on the same hardware. 

Not sure why the farm setup would be guilty here...

foosel, 2022-02-11:
I'm not saying the farm setup is guilty, I'm saying *something* is guilty, but there are too many factors at play here yet to call this an Octo**Print** bug, which this is the tracker for. You said you need help, you can get help, in the community forums, that's what they are there for. Why there and not here? Because there are way more people *there* who might be able to contribute ideas, whereas this bug tracker *here* only gets monitored by a way smaller group, most of which already has its hands full with everything else that isn't support.

bolsoncerrado, 2022-02-11:
Gotcha, thanks.