OctoPrint/OctoPrint #4417: register_custom_events will register an event with double underline...

NilsRo, 2022-02-12:
### The problem

Registering an custom event will create an constant with double underline.
[image]
 
This is not aligned with the documentation so perhaps there is an issue which will made problems after correction as the custom event id will change. Perhaps it is a wrong documentation...

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Plugins are disabled

### Version of OctoPrint

1.7.2 and also 1.7.3

### Operating system running OctoPrint

Raspian (OctoPI) and Windows 11 for Development

### Printer model & used firmware incl. version

-

### Browser and version of browser, operating system running browser

Windows 11, Firefox 97

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220212125159.zip](https://github.com/OctoPrint/OctoPrint/files/8053295/octoprint-systeminfo-20220212125159.zip)

Documentation: https://docs.octoprint.org/en/master/plugins/hooks.html#sec-plugins-hook-events-register-custom-events
[image]

jneilliii, 2022-02-12:
Is this happening because you aren't doing it within your plugin's class module? I notice you're using `"octoprint.events.register_custom_events": _register_custom_events` and the def is outside your `__pugin_implementation__`. This is how I do it...`"octoprint.events.register_custom_events": __plugin_implementation__.register_custom_events,` and I don't think there's an extra underscore.

NilsRo, 2022-02-12:
> Is this happening because you aren't doing it within your plugin's class module? I notice you're using `"octoprint.events.register_custom_events": _register_custom_events` and the def is outside your `__plugin_implementation__`. This is how I do it...`"octoprint.events.register_custom_events": __plugin_implementation__.register_custom_events,` and I don't think there's an extra underscore. And this is how I'm calling it. https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/4f893e235ec58070c4f17bf9dd7833f90610bda2/octoprint_bedlevelvisualizer/__init__.py#L531

I tried both with same result. Started like BedLevelVisualizer and changed it to outside class module because the included plugins are build like this.

https://github.com/OctoPrint/OctoPrint/blob/e5faa1985bdb4aba45048f8de49690a812439f49/src/octoprint/plugins/pluginmanager/__init__.py#L2144

jneilliii, 2022-02-12:
I'm not sure if that's intended or not, but it is matching the regex that's defined. Maybe it has to do with the plugin identifier `SlicerEstimator`, `slicerEstimator`, and `slicerestimator` are technically not the same and can all be registered in the same system?

[image]

NilsRo, 2022-02-13:
But is seems only the case for the first uppercase. What is about plugin_SliCerEstimator.   :-) It is not a 100% function then...

foosel, 2022-02-13:
This is a bug indeed

foosel, 2022-02-16:
Fixed on `maintenance`, ready for 1.8.0.

The regex was faulty. Also added some unit tests for this.

commit 3cce49fe (2022-02-16): 🐛 Fix plugin event identifier generation

plugin_Example_event was wrongly turned into
PLUGIN__EXAMPLE_EVENT. Fixed regex to solve this.

Closes #4417
closed 2022-04-05 by NilsRo