OctoPrint/OctoPrint #4418: Personalizing the user on octopi with RPI-Imager breaks the octoprint

LeNouveau, 2022-02-13:
### The problem

(I know this is for OctoPi and not OctoPrint, but I did not find the appropriate location)

# Steps 

- Create a new SD card with [RPI Imager](https://www.raspberrypi.com/software/)
- Select the OctoPi image. 
- In advanced options, activate SSH and personnalise the username. 
- Confirm all and flash the SD card. 

# Results 

On startup, the service doesn't run. 
It looks for files in `/home/pi/` but it doesn't exist. 

``` bash 
$ ps -ef | grep -i octoprint | grep -i python
$
```

``` bash 
$ sudo service octoprint status
● octoprint.service - The snappy web interface for your 3D printer
   Loaded: loaded (/etc/systemd/system/octoprint.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-12-02 13:56:28 CET; 1 years 2 months ago
 Main PID: 319 (code=exited, status=217/USER)

Dec 02 13:56:28 octopi systemd[1]: Started The snappy web interface for your 3D printer.
Dec 02 13:56:28 octopi systemd[319]: octoprint.service: Failed to determine user credentials: No such process
Dec 02 13:56:28 octopi systemd[319]: octoprint.service: Failed at step USER spawning /home/pi/oprint/bin/octoprint: No such process
Dec 02 13:56:28 octopi systemd[1]: octoprint.service: Main process exited, code=exited, status=217/USER
Dec 02 13:56:28 octopi systemd[1]: octoprint.service: Failed with result 'exit-code'.
``` 

``` bash 
$ ls /home/
myusername
```

# Suggestions 

- Update the scripts to match the configured user ? (probably complicated)
- Disable the possibility to change the username in Rpi-Imager ? (don't know if it's possible)

Regards, 
LeNouveau

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

OctoPi boot

### Version of OctoPrint

Octopi 0.18.0 with OctoPrint 1.7.3

### Operating system running OctoPrint

Raspberry

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

LeNouveau, 2022-02-13:
I can't start the application, I can't generate the export file. 
Nvm, it will be closed. ^^'

jneilliii, 2022-02-14:
That's why the download page specifically calls out to not change that. 

https://octoprint.org/download/

closed 2022-02-14 by jneilliii

LeNouveau, 2022-02-14:
I admit I did not use your documentation, as I'm used to Rpi Imager.
Sorry for the false alert.

cp2004, 2022-02-14:
>  Disable the possibility to change the username in Rpi-Imager ? (don't know if it's possible)
> > Impossible.

I think *something* is going to have to be done about this somewhere. Whether it is a request to the RPi imager people to give us the option to disable it, or a script in OctoPi that defends against the imager changing the password or something.

There's one or two posts on the forums, as well as a couple on the discord server. Because it is easy/intuitive to install through the imager, people don't read the rest of the download page, or they are following a non-official resource.

foosel, 2022-02-14:
The only way we could get the imager to not allow to change the use rname (the password change is actually welcome) is by completely disallowing the imager to change anything. Which also means wifi config. The mechanism for that involves installing a shell script that runs as init process. No way around it, it starts before anything else starts. 

So a feature request for the pi people it should be I guess. Personally I'd prefer if we could not just say yes or no to customization at all and instead also have some more granular control, but currently that's not there, and I considered it more important to allow UI support for wifi config and hoping people would read and follow setup instructions, instead of disallowing this functionality altogether. If someone would like to open (or add to) a public feature request to rpi imager, that would be very welcome. I can also reach out internally, but that has better chances if there's some public interest documented beyond just me.