OctoPrint/OctoPrint #4419: [Request]

Dirki74, 2022-02-14:
### Is your feature request related to a problem? Please describe.

If the webcam is not turned on, a password may be displayed if a password is required to connect to dr Webcan. In many cases this is a security problem. 

### Describe the solution you'd like

It would be nice if the webcam URL could be hidden.

fieldOfView, 2022-02-14:
The webcam url is not a feature of OctoPrint, but of OctoPi. Your request is a duplicate of https://github.com/guysoft/OctoPi/issues/292

closed 2022-02-14 by fieldOfView

cp2004, 2022-02-14:
If a password is embedded in the webcam URL, then there is no way to hide it. It will be visible in the browser's network requests list, if someone presses 'inspect' in the browser they would see it as well. In order to display the webcam stream the URL would have to be in the settings API, again plainly visible to everyone. There's no way to securely hide the URL.

This is also the reason a lot of browsers now block embedding credentials in the URL for basic authentication.

fieldOfView, 2022-02-14:
The only way this could be solved is if (on the OctoPi side of things) HAProxy would somehow be able to access the session cookie set by OctoPrint and derive permissions from that. But that would be hard to implement without opening up this mechanism to session cookie spoofing.