OctoPrint/OctoPrint #4421: [1.8.0dev] Restart print button is always disabled

cp2004, 2022-02-15:
### The problem

1. Start print
2. Pause print
3. Try and restart - notice restart button is disabled.

Occurs as a result of #4336 - before this PR it works, after it does not.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0.dev

### Operating system running OctoPrint

Windows 11

### Printer model & used firmware incl. version

VIRTUAL

### Browser and version of browser, operating system running browser

Edge 98

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220215212413.zip](https://github.com/OctoPrint/OctoPrint/files/8074845/octoprint-systeminfo-20220215212413.zip)

jneilliii, 2022-02-15:
I can confirm this bug in latest maintenance branch from this morning.

foosel, 2022-02-16:
Yep, can confirm as well.

foosel, 2022-02-16:
Fixed by the above commit.

commit 82338209 (2022-02-16): 🐛 Enable print button if ready OR paused

Closes #4421