OctoPrint/OctoPrint #4422: Octoprint timeout on prusa firmware 3.10.1 with MMU2S

knobik, 2022-02-16:
### The problem

i get timeout messages, cuz prusa firmware 3.10.1 does not send the `busy: processing` anymore  and after that point octoprint sends `M112` as it times out.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

octopi 0.18.0

### Printer model & used firmware incl. version

prusa i3 mk3s+ (3.10.1) with mmu2s (1.0.6)

### Browser and version of browser, operating system running browser

latest chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220216115908.zip](https://github.com/OctoPrint/OctoPrint/files/8079261/octoprint-systeminfo-20220216115908.zip)

```
Send: N16 M190 S85*98
Recv: T:104.8 /240.0 B:49.1 /85.0 T0:104.8 /240.0 @:127 B@:127 P:0.0 A:34.3
Recv: echo:busy: paused for user
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: T:104.5 /240.0 B:49.0 /85.0 T0:104.5 /240.0 @:127 B@:127 P:0.0 A:34.5
Recv: mmu_get_response - begin move: T-code
Recv: MMU <= 'T0'
Recv: echo:busy: processing
Recv: Unloading finished 2
Recv: T:105.0 /240.0 B:48.9 /85.0 T0:105.0 /240.0 @:127 B@:127 P:0.0 A:34.3
Recv: echo:busy: processing
Recv: mmu_get_response - begin move: load
Recv: T:106.9 /240.0 B:48.9 /85.0 T0:106.9 /240.0 @:127 B@:127 P:0.0 A:34.5
Recv: echo:busy: processing
Recv: T:109.7 /240.0 B:49.1 /85.0 T0:109.7 /240.0 @:127 B@:127 P:0.0 A:34.5
Recv: echo:busy: processing
Recv: T:113.9 /240.0 B:49.3 /85.0 T0:113.9 /240.0 @:127 B@:127 P:0.0 A:34.7
Recv: echo:busy: processing
Recv: T:118.6 /240.0 B:49.9 /85.0 T0:118.6 /240.0 @:127 B@:127 P:0.0 A:34.7
Recv: echo:busy: processing
Recv: T:123.8 /240.0 B:50.2 /85.0 T0:123.8 /240.0 @:127 B@:127 P:0.0 A:34.9
Recv: echo:busy: processing
Recv: T:128.9 /240.0 B:50.8 /85.0 T0:128.9 /240.0 @:127 B@:127 P:0.0 A:35.0
Recv: echo:busy: processing
Recv: T:133.4 /240.0 B:51.4 /85.0 T0:133.4 /240.0 @:127 B@:127 P:0.0 A:35.1
Recv: echo:busy: processing
Recv: T:138.7 /240.0 B:52.2 /85.0 T0:138.7 /240.0 @:127 B@:127 P:0.0 A:35.4
Recv: echo:busy: processing
Recv: T:144.1 /240.0 B:53.0 /85.0 T0:144.1 /240.0 @:127 B@:127 P:0.0 A:35.5
Recv: echo:busy: processing
Recv: T:149.3 /240.0 B:53.6 /85.0 T0:149.3 /240.0 @:127 B@:127 P:0.0 A:35.4
Recv: echo:busy: processing
Recv: T:154.6 /240.0 B:54.4 /85.0 T0:154.6 /240.0 @:127 B@:127 P:0.0 A:35.4
Recv: echo:busy: processing
[... 244 lines cut ...]
Recv: T:240.94 E:0 B:83.5
Recv: T:240.52 E:0 B:83.6
Recv: T:240.4 /240.0 B:83.6 /85.0 T0:240.4 /240.0 @:23 B@:63 P:0.0 A:38.7
Recv: T:240.63 E:0 B:83.7
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N46 M105*21
Recv: T:240.00 E:0 B:83.7
Recv: T:240.4 /240.0 B:83.8 /85.0 T0:240.4 /240.0 @:24 B@:63 P:0.0 A:38.9
Recv: T:240.31 E:0 B:83.8
Recv: T:240.10 E:0 B:83.9
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N47 M105*20
Recv: T:240.2 /240.0 B:84.0 /85.0 T0:240.2 /240.0 @:26 B@:72 P:0.0 A:38.7
Recv: T:240.31 E:0 B:83.9
Recv: T:239.69 E:0 B:84.1
Recv: T:239.8 /240.0 B:84.0 /85.0 T0:239.8 /240.0 @:31 B@:83 P:0.0 A:38.8
Recv: T:239.30 E:0 B:84.1
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N48 M105*27
Recv: T:239.61 E:0 B:84.1
Recv: T:239.5 /240.0 B:84.1 /85.0 T0:239.5 /240.0 @:35 B@:88 P:0.0 A:38.8
Recv: T:238.75 E:0 B:84.2
Recv: T:238.59 E:0 B:84.2
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
```

foosel, 2022-02-16:
> cuz prusa firmware 3.10.1 does not send the `busy: processing` anymore and after that point octoprint sends `M112` as it times out.

Then please report this to Prusa and not OctoPrint. What is OctoPrint supposed to assume but the printer going AWOL if it's not responding anymore for lengthy periods of time? That's precisely what the `busy` protocol is supposed to solve.

closed 2022-02-16 by foosel

foosel, 2022-02-16:
That is because `Tx` is not a valid GCODE command but something Prusa or rather Prusa MMU specific. OctoPrint does only send checksums and line numbers for lines it recognizes as valid GCODE lines, as a there exist firmware variants out there that support out of band messages that must not be checksumed, which then malfunction on checksums on these commands. The same goes for waiting for an acknowledgement. You can change this behaviour, however the settings for that are currently not exposed in the UI and needs you to manually [edit config.yaml](https://docs.octoprint.org/en/master/configuration/config_yaml.html) and set `serial.sendChecksumWithUnknownCommands` and `serial.unknownCommandsNeedAck` to `true`, e.g.

``` yaml
serial:
  sendChecksumWithUnknownCommands: true
  unknownCommandsNeedAck: true
```

In any case, even in heat-up the printer is expected to produce `busy` messages once it states it supports them (by producing any). It doesn't here, and you also state this started with a new firmware version, whereas the behaviour regarding `Tx` in OctoPrint has been like this forever, so this still looks like something that Prusa changed and now need to fix on their side. Either the firmware supports the `busy` protocol for ongoing processes (long moves, heatups, etc) or while waiting for user interaction (there's a `busy:waiting for user` there, so support is assumed), or it doesn't. Half measures cause issues like this one.