OctoPrint/OctoPrint #4428: [Request] Plugin manager queue. "install later" "reboot later" option.

tastyratz, 2022-02-17:
### Is your feature request related to a problem? Please describe.

My printer is always printing! I'd love to make changes while it's printing instead of waiting till later. Also, sometimes I want to install/uninstall several plugins in 1 go, but, I need to reboot for each change.

### Describe the solution you'd like

I understand making live changes during a print is ill-advised. What if changes to plugins could be queued till when octoprint is not busy? They can install after the current job/task is completed.

Also, I understand the reason you might want to watch the plugin install shell and reboot after it completes, but, what if that could happen in the background and the reboot would be required before closing the settings/plugin menu or before the next print instead?

This way I could just queue up my changes, fire them off when ready, and reboot when tasks are complete without having to sit at the terminal.

### Additional context

Also, the plugin manager page in settings is in a bit of an awkward location and the options are in no particular sort (instead of alphabetical). 
Maybe it makes sense to move to the "features" menu or pin to the top of the "plugins" list?

jneilliii, 2022-02-19:
I've implemented a software update queue in the above PR, hopefully to be part of 1.8.0. This does not cover plugin manager installs but is a step in the right direction. Won't have time myself to implement this, but I bet the changes would be fairly similar.

jneilliii, 2022-08-07:
I'd be willing to do the enqueuing of plugin installs while printing similar to the software update, but think there will probably be conflicts associated with #4478, which seems to have stalled. @foosel, would it be appropriate/acceptable to pull those changes into my dev and address comments and submit along with my PR for enqueuing plugin installs?

foosel, 2022-08-08:
@jneilliii absolutely, go for it!

closed 2024-05-22 by cp2004