OctoPrint/OctoPrint #4431: [Request] Pending restart after plugin update

RoboMagus, 2022-02-18:
### Is your feature request related to a problem? Please describe.

When installing plugins / manually updating them one by one, Octoprint automatically restarts.
When having to perform multiple actions that require a restart of Octoprint it would be nice to do them all in one go, instead of having to reboot in between.

### Describe the solution you'd like

It would be nice to have updating a plugin set some sort of `pending restart` flag that would allow users to perform other tasks before the restart is actually issued.

How this can be implemented practically is up for debate. Either have a notification message with a timeout after which the restart occurs, or have it restart on the next disconnect of the printer. Have a nice big restart button in the navbar to draw the user's attention to do it themselves later, etc...

Either implementation, I think it would at least be wise to prevent users from starting a new print while this pending flag is set. To prevent too long postponing. (Lets all remember windows updates...)

### Describe alternatives you've considered

Stopping my Octoprint docker container and updating / installing plugins from the CLI.
Or just waiting for each restart in between my manual update actions.

RoboMagus, 2022-02-18:
I was triggered for this request after reading #4428. But that seemed to be a different use-case. (Modifications to be postponed till prints are done)

jneilliii, 2022-02-19:
I've implemented that in the above PR and awaiting approval to hopefully be included in 1.8.0.

closed 2022-02-19 by jneilliii