OctoPrint/OctoPrint #4434: [Request] Gcode Viewer support for CNC machines (which have no extrusion)

vector76, 2022-02-20:
### Is your feature request related to a problem? Please describe.

In many cases, gcode generated by a CAM processor for a CNC machine does not show properly in the GCode Viewer.

As an example, when following these instructions:
https://www.v1engineering.com/estlcam-basics/
which refer to using this gcode file:
https://www.v1engineering.com/wp-content/uploads/2015/09/Test-Crown-12mms.gcode
nothing is rendered in the GCode viewer.

Since there are no "E" movements in the gcode file, the viewer has nothing to render.

### Describe the solution you'd like

Ideally I would prefer a settings option within the built-in GCode Viewer to enable "CNC Mode", which uses Z height to infer which segments are cutting operations and which segments are travel moves.

### Describe alternatives you've considered

I have built a plugin which is a clone of the built-in GCode Viewer, with a few small changes to render cutting moves and travel moves as such.  The github repo is here: https://github.com/vector76/cnc_gcodeviewer and this commit in particular contains all the meaningful changes: https://github.com/vector76/cnc_gcodeviewer/commit/2afeef2cec06ed253c620b049dabef7e574bbd9c 

If "CNC Mode" option is not added to Octoprint, I can package the modified GCode Viewer and submit it to the plugin repository, which is equally good in my opinion but I wanted to ask if it belonged in the upstream Octoprint.

jneilliii, 2022-02-20:
There have been recent pull requests related to gcode viewer with performance improvements. Not sure if it makes sense to incorporate this as a core option, but might make sense to bundle your viewer as a replaces style view model in a plugin. Potentially integrated into the grbl plugin?

vector76, 2022-02-20:
I am using Marlin, so it would at least be orthogonal to grbl.  I don't know anything about the grbl plugin or if it has its own working viewer.

Constructing a replaces style view model plugin is probably beyond what I am willing to invest in learning.  Replacing the entire GCode Viewer with a small change has its own downsides, but it is easy and functional, so I think that is still my second choice.

I think I could create a PR that adds the 'CNC Mode' as described, but I wanted to first gauge if there was an appetite for such a thing, or if that use-case is too much of an outlier to warrant an option.

foosel, 2022-02-21:
I'd consider this more suited for a plugin than a core change. The reason is that OctoPrint primarily targets printers, not CNCs, the majority of its users has printers, not CNCs, and in general the goal is to move any kind of development and maintenance and support overhead out of core that doesn't benefit the huge majority of the user base, simply to keep things better manageable and to better spread the burden of maintenance across the interested parties in the community.

vector76, 2022-02-23:
Thank you for the feedback.  I can appreciate that too many options, if they are rarely used, can degrade the primary use-case.

The plugin repository is just a couple clicks away, and I think it is an equally good solution for those that need it.

closed 2022-02-23 by vector76