OctoPrint/OctoPrint #4436: [Request] add a UI button to swap between Octoprint instances (IPs) in the same network

LidorBaum, 2022-02-22:
### Is your feature request related to a problem? Please describe.

I have 2 Raspberri Pis running Octoprint on 2 printers. I am willing to have a nice option to swap between them in browser, instead of writing the IP in the URL.

### Describe the solution you'd like

If you know the WLED library, there is also a button to swap between instances of WLED in the same network by IP address, very easy and intuitive.

OllisGit, 2022-03-18:
Hi @LidorBaum, 
do you mean something like this:
[image]
[image]

Maybe I need an additional option "open in new tab"...will see.

LidorBaum, 2022-03-21:
@OllisGit Hi, exactly this! I can not find it in the octoprint plugins, nor in your repo (its empty), is it released yet?

fieldOfView, 2022-03-21:
> Maybe I need an additional option "open in new tab"...will see.

I would do that with the normal CTRL+click.

What could be an interesting feature is to use bonjour discovery to populate the list automatically.

OllisGit, 2022-03-22:
Hi,
I change the plugin name to "BookmarkManager" and released the first version.

https://github.com/OllisGit/OctoPrint-BookmarkManager

It now looks like this:
[image]

PS. the  "bonjour discovery" feature is also tracked in the repo ;-)

closed 2024-08-29 by jneilliii