OctoPrint/OctoPrint #4440: Pause not populating pause_position after first connection to serial

mscreations, 2022-02-27:
### The problem

I'm having an issue with pausing. I am using PrusaSlicer to add a pause at a certain height. It added an M601 which my printer does not support, however, I have configured Octoprint to pause on M601. I then have a GCODE script that is a simplified version of the one from OctoPrint's documentation (it only moves the extruder. No retraction or extruder movement). In the GCODE script for "after a print job is paused", it checks if pause_position.x is None and if not moves the extruder to the back left. The "before print job is resumed" restores the extruder position. 

I am finding that after the first time I connect serial to the printer, pause_position is None and my pause script doesn't execute. Looking at the logs it appears as though the  PrintPaused event is being fired before all the moves are complete and it executes an M114 to get the paused position. It therefore does not have valid position data for the pause. I am not sure why it works correctly the second time after the serial is connected. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Biqu B1 Marlin 2.0.9.3x

### Browser and version of browser, operating system running browser

Opera 83.0.4254.46

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220227052402.zip](https://github.com/OctoPrint/OctoPrint/files/8148089/octoprint-systeminfo-20220227052402.zip)
[test.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/8148090/test.gcode.txt)

cp2004, 2022-02-27:
I decided this looked like a bug to me after looking through the serial.log and confirming M114 commands were sent and answered. One possible theory based on that was that the M114 cmd/response appears before it says 'Changing monitoring state...', so something is out of order

mscreations, 2022-02-28:
I did have a manual M114 before the M601 code from when I was troubleshooting. I did confirm though that if I change the M601 to a M400 and THEN M601 the printer works as expected everytime. I just can't understand why the issue is ONLY when I try to pause after first connecting to the printer. I discovered this as an issue when I was trying to run a print and paused to insert a lens into place (after an hour of printing). The extruder head stopped right over where I needed to put it in even though it should have parked it. I created the test code with just a few moves to see if it was a problem with the actual sliced file or what and was able to limit it to only after first connection.

foosel, 2022-03-02:
I actually don't think that there's anything out of order here, but rather your printer's command buffer is too large and thus the 10s timeout for waiting for the position report on a pause (and likely also cancel) is too small. Try increasing it to 40s (even 30 might still be too small judging by your log): Settings > Serial Connection > Intervals & Timeouts > Timeouts > Position query timeout

To elaborate, the relevant part in the code is here:

https://github.com/OctoPrint/OctoPrint/blob/e3e2d4dbeb7078e08325ebb46fd4a89f5cd8c574/src/octoprint/util/comm.py#L1890-L1946

So, what happens in case of a pause is, first it checks if autoreporting is on, and if so puts a `M114 S0` in the job command queue to disable it. Then it enqueues an `M400` to wait for all movement to finish. After this `M400` is sent, a callback enqueues an `M114` and starts a timer waiting for the result from that. The `M114` upon being sent will also enqueue an `M114 S<interval>` with the old autoreporting interval, if needed.

All of these commands go into the job queue in this order:
 
  1. `M154 S0`
  2. `M400`
  3. `M114`
  4. `M154 S<interval>`

What we see here, repeatedly I might add, is this:

  1. `M114` (no idea where that's coming from)
  2. state = pausing
  3. `M154 S0`
  4. `M400`
  7. *loooooong wait, around 30s, while the moves already sent are executed*
  8. `M114`
  9. `M154 S<interval>`
  10. state = paused

As the execution of the `M400` takes so extremely long, during 7 a timeout is triggered and position recording is given up, pausing script is prepared and already enqueued. It then gets sent after `M154`, at which point the pause position actually DOES get recorded (which funnily enough is a bug) and then get used later (definitely a bug), but the pausing script has already been processed and enqueued *without* the pause position.

So, tldr, increase the position query timeout to 40s, see if that helps. Either way, please report back.

mscreations, 2022-03-04:
So the M114 in your list above as #1 is from when I was troubleshooting. It was in the gcode file before the M601.

 I increased the position query timeout and that seems to have fixed the issue. Thank you for all your help! I will close this as it appears to be solved on my end.

closed 2022-03-04 by mscreations