OctoPrint/OctoPrint #4447: Rendering of Timelapse failed with return code -9

muckypaws, 2022-03-07:
### The problem

1) Enable Timelapse in OctoPi
2) Run a print
3) When Print is complete, message appears to advise the Timelapse is rendering.
4) CPU 1 goes 100% and Raspberry Pi becomes unresponsive for a while
5) Timelapse fails with error code -9

ffmpeg version 4.3.3-0+rpt2+deb11u1 Copyright (c) 2000-2021 the FFmpeg developers
,  built with gcc 10 (Raspbian 10.2.1-6+rpi1)
,  configuration: --prefix=/usr --extra-version=0+rpt2+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
,  WARNING: library configuration mismatch
,  avutil      configuration: --prefix=/usr --extra-version=0+rpt2+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
,  avcodec     configuration: --prefix=/usr --extra-version=0+rpt2+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
,  avformat    configuration: --prefix=/usr --extra-version=0+rpt2+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
,  avdevice    configuration: --prefix=/usr --extra-version=0+rpt2+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
,  avfilter    configuration: --prefix=/usr --extra-version=0+rpt2+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enab [... cut ...]
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]
[octoprint-systeminfo-20220307201203.zip](https://github.com/OctoPrint/OctoPrint/files/8200557/octoprint-systeminfo-20220307201203.zip)

deftdawg, 2022-04-23:
Same problem here, also on a Raspberry Pi Zero W 2.

OctoPrint: `Version 1.7.3`
OctoPi: `Version 1.0.0, running on Raspberry Pi Zero 2 W Rev 1.0`

I have successfully rendered shorter prints but this particular print was 14 hours with 401 snapshots in `~/.octoprint/timelapse/tmp`

Wonder if the error -9 is from running out of RAM or something.

In any case, Octoprint did not catch the error and the UI is stuck at 0%:

[image]

though it does show the error in the log

`grep octoprint.timelapse ~/.octoprint/logs/octoprint.log`

2022-04-23 16:55:14,963 - octoprint.timelapse - WARNING - Could not render movie, got return code -9 (reformatted below):
```
ffmpeg version 4.3.3-0+rpt3+deb11u1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10 (Raspbian 10.2.1-6+rpi1)
  configuration: --prefix=/usr --extra-version=0+rpt3+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version=0+rpt3+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version=0+rpt3+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-mmal --enable-neon --enable-rpi --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version=0+rpt3+deb11u1 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-gnutls --enable-ladspa --enable-libaom --enable-lib [... cut ...]     0kB time=-00:00:00.07 bitrate=N/A speed=N/A    \r', 'frame=   43 fps=1.6 q=24.0 size=     256kB time=-00:00:00.03 bitrate=N/A speed=N/A    \r', 'frame=   44 fps=1.0 q=28.0 size=     256kB time=00:00:00.00 bitrate=26891487.2kbits/s speed=1.74e-06x    \r']
```

Probably the next step is to figure out the commandline that OctoPrint using to run ffmpeg and see if it works successfully when executed directly (or if it works with a subset of the snapshots to see if it's memory constraint related).

deftdawg, 2022-04-23:
Another thing I'm going to do as well is go into Raspi-config and reduce the vram, stock RPi ZW2 is using 128M for vram and that means it only has 384M for system ram.

`sudo raspi-config`
`Performance -> P2 GPU Memory`

And reduce the video RAM allocation from `128MB` to `16MB`, save and reboot.

deftdawg, 2022-04-23:
Here's how to re-run the job:

```sh
export JOB_NAME=AET4PRO_Bdstesla_cable_holder_20220423023319
export JOB_STATUS="-fail"
/usr/bin/ffmpeg -r 25 -i "/home/pi/.octoprint/timelapse/tmp/${JOB_NAME}-%d.jpg" -vcodec libx264 -threads 1 -b:v 10000k -f mp4 -y -vf '[in] format=yuv420p [postprocessed]; movie=/home/pi/oprint/lib/python3.9/site-packages/octoprint/static/img/watermark.png [wm]; [postprocessed][wm] overlay=10:main_h-overlay_h-10 [out]' "/home/pi/.octoprint/timelapse/.${JOB_NAME}${JOB_STATUS}.mp4"
```

The pi pretty much crashed after trying this... lol

Time to apply the Ram change and retry it.

deftdawg, 2022-04-24:
Apparently one must disable Pi Camera support by removing `start_x=1` from `/boot/config.txt`, otherwise 128M will always be reserved for video, regardless of the raspi-config `P2 GPU Memory` setting (`raspi-config -> Interface Options-> I1 Legacy Camera Support -> Dislable` and reboot).

cp2004, 2022-04-29:
So is your conclusion @deftdawg that it is a memory related issue? Or is there more investigation to go?

Even if the rendering error is not solvable from OctoPrint's end, the issue with it showing stuck at 0% should be.

deftdawg, 2022-04-30:
With the `start_x=1` change to `/boot/config.txt` I get 475MB of free ram (w/ `gpu_mem=16`).   I was able to manually render some failed Timelapse's...  The larger of the two was 71 frames, and that used all of the available ram, and 75MB of the 100MB of swap.

I ran them inside of byobu, and it seemed like the process was only really working on a single core. 

My best guess is that it's memory constrained, and trying to render something that exceeds mem+swap will make the system become unresponsive.

Overall, considering I can't actually buy a Pi4 for less than double it's MSRP, I'm pretty happy with the Pi 2 Zero W, even if it can't handle large timelapses.

foosel, 2022-05-11:
Don't see anything that OctoPrint could be doing here tbh, apart from maybe warning about large timelapses potentially being unrenderable, but that would require a new feature request if that is something people are interested in (and require some data collection to figure out a sensible threshold). 

Closing as not immediately actionable and outside of OctoPrint itself.

closed 2022-05-11 by foosel

muckypaws, 2022-05-12:
Hi @foosel Have to disagree here, and would like the issue re-opening.  I don't believe the issue is with large renders, as I can recreate the issue with a short/cancelled print that consists of just a few frames.  I suspect an issue with the command used to stitch together the still frames into an MP4 that's at fault.  Could be a bug with FFMPEG or the command line arguments passed via OctoPrint.

foosel, 2022-05-12:
It is ffmpeg that is returning an error here, after lengthy processing. What do you expect me as the maintainer of *OctoPrint* to do about that? The parameters can be modified, I cannot reproduce the issue on any of my machines and the only reproduction we had from someone else was - to my understanding - identified as memory related.

Watt25, 2023-06-23:
Increase Swap size.
sudo dphys-swapfile swapoff
sudo nano /etc/dphys-swapfile
Modify CONF_SWAPSIZE to 512
(Save file.)
sudo dphys-swapfile setup
sudo dphys-swapfile swapon

thomaslazar, 2024-01-14:
Hi. I encountered this exact issue as well now on a Raspi Zero 2 today. And the solution from @Watt25 solved this problem. In general I wonder why there's only 100MB swap size configured on the octopi image. Is this something to be changed in this repository or do I need to make a ticket for that someplace else?

jneilliii, 2024-01-14:
Swapsize would be an OctoPi setting, so maybe on that repo. https://github.com/guysoft/OctoPi

nistvan86, 2024-07-22:
> Increase Swap size.

This is not really a solution. It relies heavily on the swap file, which going to kill the SD card in no time.

I would rather finetune the ffmpeg command line, eg. don't try to encode it into H.264 on the Pi Zero 2W, just mux mjpeg into some kind of container which can be played with media players (I think mkv should work), and I'll handle the transcode on my desktop PC if it's worthwhile to do.

Another option which could be more resource friendly is compiling an ffmpeg version which supports the `-c:v h264_v4l2m2m` hardware encoder of the Pi, instead of doing it on the CPU with x264. (probably more trouble than worth it)