OctoPrint/OctoPrint #4448: Show the Part cooling Fan speeds on temp Graph ...

jseaz73, 2022-03-09:
### Is your feature request related to a problem? Please describe.

It is helpful to know why there are changes in the temp ... I have found, lately, that just walking up on the printer caused a change to the hot end temp ... 

### Describe the solution you'd like

have the controls for the fan %  and add a graph over time for the fan percentage on the default Temp tab.

### Describe alternatives you've considered

I have tried looking in the plugin Lib no good options

jneilliii, 2022-03-09:
My plotly temp graph plugin can do that with an additional supplemental plugin. Fan speed was actually one I used as a test.

jseaz73, 2022-03-09:
> I do not see how I can put Fan Speed % on the graph

jseaz73, 2022-03-09:
> I do not see how I can put Fan Speed % on the graph
>
> You have to also install this file in plugin manager.
>
>
> https://raw.githubusercontent.com/jneilliii/OctoPrint-PlotlyTempGraph/master/test_plotly_graph_fan_speed.py

Forgive my ignorance here.  I have added both and I still am not seeing fan
speed on the chart.  M106 S### does change the fan speed
but I do not see it on the chart

I have added both

jneilliii, 2022-03-09:
That means your firmware isn't reporting/echoing the change. I do think that was an optional setting in marlin that defaults off. Would require changes to the single file plugin.

jseaz73, 2022-03-09:
Thank you for the help.  Do you know which feature in Marlin ?  I just
updated to 2.0.9.3

I am comfortable with making changes to the Firmware

jneilliii, 2022-03-10:
REPORT_FAN_CHANGE

hayden-t, 2022-06-30:
hi this would be a good option, i can see my fan speed with top temp bar plugin, but would like it graphed too

alsmithson, 2023-11-18:
I am troubleshooting some ABS prints that require fan in some places. I would very much like to have the fan speed graphed so I don't have to sit and watch the dashboard fan gauge through the whole print. Thank you for your consideration.