OctoPrint/OctoPrint #4453: [1.8.0rc1] HTTP 500 after update

foosel, 2022-03-14:
#### What were you doing?

1. update to 1.8.0
2. after reboot web GUI give me a 500 error

#### What did you expect to happen?
a successful upgrade
#### What happened instead?
an unsuccessful upgrade and a 500 error on the web GUI
#### Did the same happen when running OctoPrint in safe mode?

Tested in safe mode and thing appear to work as expected. It is indeed a 3rd party plugin crashing it. I know you dont want plugin bugs here, but maybe with it being an RC you do.

#### Version of OctoPrint

1.8.0

#### Printer model & used firmware incl. version

Raise 3D Pro2 Plus

#### Browser and version of browser, operating system running browser

debian desktop and windows 10

#### System Info Bundle

included.

Im not sure how to figure out what plugin is crashing the non-safe mode boot.

[octoprint-systeminfo-20220314124301.zip](https://github.com/OctoPrint/OctoPrint/files/8244842/octoprint-systeminfo-20220314124301.zip)

I have read the FAQ.

---

Bundles:

  * [octoprint-systeminfo-20220314124301.zip](https://bundleviewer.octoprint.org/?url=https%3A%2F%2Fgithub.com%2FOctoPrint%2FOctoPrint%2Ffiles%2F8244842%2Foctoprint-systeminfo-20220314124301.zip)

*edited by @github-actions to add bundle viewer links*

_Originally posted by @pingywon in https://github.com/OctoPrint/OctoPrint/issues/4452#issuecomment-1066743041_

foosel, 2022-03-14:
@pingywon haven't yet figured out what is happening there in your report, never seen that before, but:

> !!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.

your Pi ran into an undervoltage issue so you need to fix that in any case, even though it wasn't active anymore after reboot, just running into that *at all* indicates an issue.

pingywon, 2022-03-14:
> @pingywon haven't yet figured out what is happening there in your report, never seen that before, but:
>
>
> your Pi ran into an under voltage issue so you need to fix that in any case, even though it [...]

I have just begun to report the under voltage. I think the PS is failing, no doubt. I have a replacement on order, just a couple days, but this install had been working fine for days before the upgrade.

Is there anyway to easily find the offending plug-in?

foosel, 2022-03-14:
Basically do a binary search. So, disable half, enable the other half, test. If it works, the offending plugin is in the disabled half, if not, in the enabled half. Half the half again, rinse and repeat until you've found the plugin. 

It's definitely something that also needs more error resilience in OctoPrint, and I'm looking into that, but it would still be nice to know what plugin is causing this specifically so I can test if my work around works.

pingywon, 2022-03-14:
> Basically do a binary search. So, disable half, enable the other half, test. If it works, the offending plugin is in the disabled half, if not, in the enabled half. Half the half again, rinse and [...]

I didnt know if there was another way. I'll try to narrow it down for us.

pingywon, 2022-03-14:
[This Plug-in](https://github.com/gruvin/OctoPrint-WebcamTab) seems to be the offender. Once removed Octo boots as expected.

pingywon, 2022-03-14:
Thanks! @foosel

foosel, 2022-03-14:
Fix ready for 1.8.0rc2

foosel, 2022-03-16:
1.8.0rc2 has been released.

commit e59fab10 (2022-03-14): 🥅 More resilience against template plugin bugs

  * Don't crash if template is set to None
  * In general catch any error while processing plugin template
    configs and ignore plugin completely.

Closes #4453
closed 2022-03-16 by foosel