OctoPrint/OctoPrint #4455: Multiple Octoprint enabled printers unable to check for updates - "Rate limit exceeded while checking for update, please try again later"

Naugrimohtar, 2022-03-14:
### The problem

I have two printers, each has a Raspi running OctoPrint. I find it frustrating that when I try to check for updates I do it on both systems at the same time and one returns with "Rate limit exceeded while checking for update, please try again later". Example happened today when I updated both to the late 1.8.0 rc1 on both systems in parallel. When both rebooted, only one showed that a plugin had an update (arcwelder). Although the plugin is installed on both systems, the other did not tell me there was an update because of "Rate limit exceeded while checking for update, please try again later".

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

many but 1.8.0rc1

### Operating system running OctoPrint

OctoPi 0.17.0

### Printer model & used firmware incl. version

Prusa MK2.5 and MK3s, latest from Prusa

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

N/
[octoprint-systeminfo-20220314090405.zip](https://github.com/OctoPrint/OctoPrint/files/8245480/octoprint-systeminfo-20220314090405.zip)
A

foosel, 2022-03-14:
This is covered in the FAQ: https://community.octoprint.org/t/i-get-something-about-rate-limit-exceeded-when-trying-to-update-octoprint-or-plugins-help/33550

foosel, 2022-03-14:
No options need to be checked or the FAQ would have mentioned this. I also see that your instance was updated to 1.8.0rc1 successfully in the system bundle you shared, so please elaborate on *what update targets* you see the rate limit, which of the options in the FAQ you chose to add the token and share another system info bundle from an affected instance.

Naugrimohtar, 2022-03-14:
Below are system info bundles from both of my instances.
Also are screenshots from the update page.
After a few hours of leaving these along, I still am seeing "Rate limit exceeded while checking for update, please try again later"

Of specific note, compare screenshot status for the 'Arc Welder' plugin and its version.

[octoprint-systeminfo-20220314135553_MK2.5.zip](https://github.com/OctoPrint/OctoPrint/files/8247826/octoprint-systeminfo-20220314135553_MK2.5.zip)
[image]
[octoprint-systeminfo-20220314135641_MK3s.zip](https://github.com/OctoPrint/OctoPrint/files/8247829/octoprint-systeminfo-20220314135641_MK3s.zip)
[image]

I used the "OctoPrint 1.8.0 and later" approach per the FAQ

foosel, 2022-03-14:
As far as I can see the Arc Welder plugin in version 1.0.0 is broken because there it will not include the current version in the update configuration in OctoPrint 1.5.0+, which explains the broken display in the second screen shot. The first one is on 1.1.0rc2 and that has the config fixed.

Does the rate limit on the other items persist if you force check for updates?

[image]

Naugrimohtar, 2022-03-15:
Yes it does persist, even after hours between attempts.

foosel, 2022-03-15:
Just to verify, even with a force check for updates, so the button above, not the regular "Check for update" one?

Naugrimohtar, 2022-03-15:
Correct, yesterday it was showing persistent rate limit even with 'force update'.

Today, I see it has cleared and seems to be checking for updates correctly.

I have not changed anything more on my end other than another day elapsing. Perhaps it took time for the github side of things to support the configuration change.

foosel, 2022-03-16:
Pretty sure this was caused by #4458 which has been fixed in 1.8.0rc2 and would have let to the token to be ignored in 1.8.0rc1. Closing this as 1.8.0rc2 is now available and if push comes to shove there's also [hotfix plugin](https://gist.github.com/foosel/42bec0a3976a676a4e98d3ac59ac3f22).

closed 2022-03-16 by foosel

STekZeNIO, 2022-04-07:
The fix that did it successfully for me was:

1.  [Create a personal access token](https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/creating-a-personal-access-token)
2.  [Edit config.yaml](https://community.octoprint.org/t/cannot-update-rate-limit-exceeded/30258/28)
  `softwareupdate:
    credentials:
      github: <your github personal access token goes here>`
3.  [Force check for updates](https://github.com/OctoPrint/OctoPrint/issues/4455#issuecomment-1067221245)

That's it!