OctoPrint/OctoPrint #4457: [RC Feedback] Feedback on the 1.8.0rc2 Release Candidate

foosel, 2022-03-15:
Please provide *general* feedback on your experience with the 1.8.0rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

pingywon, 2022-03-15:
Has it been released? Im not seeing it.

foosel, 2022-03-15:
No, it hasn't yet. I prepared the ticket halfway through the initial update test, then ran into #4458 and since then have been working on this and #4459 now as well. Sometimes stuff doesn't run as straightforward as planned, you'll see once it's released by it actually showing up under https://github.com/OctoPrint/OctoPrint/releases and also a release (candidate) announcement on the blog, twitter, facebook, discord, forums, ...

pingywon, 2022-03-16:
All my issues (including my under voltage lol) appear to currently be resolved. 
I updated to rc2 with out an issue. 
I was able to install  [WebcamTab](https://github.com/gruvin/OctoPrint-WebcamTab) and after a reboot of octo and then a reboot of the whole pi I can confirm that rc2 boots fine with WebcamTab plugin installed. 

How ever, for me the plug-in did not show the webcam and actually had to be uninstalled before webcam vid returned. At this point it is def a plug-in issue. I do not need that plug-in what so ever. I just happened to have it installed on one of my octo instances.

Thanks again Gina for all the hard work.

mod38, 2022-03-16:
I have just posted a bug report (!) about another problem with the gcode viewer which is that it stops updating one layer early so on my 10 layer test piece it finishes at layer 9. Up to that it is behaving perfectly.

Cpat39, 2022-03-17:
Running 1.8 Orc 2 on Pi Zero 2 W. Have plug-ins running. Arc Welder, PreHeat Button, Pretty Gcode, TemperatureLegendMover, DashBoard and GcodeViewer  and TSD. Printer is AnyCubic Vyper. Laptop is MacBook Pro (2016) with Monterey 12.2.1. All running as expected. Model has 16 layers. None of standard Plug-ins have been deleted. Python V3.7.3. Still unable to SSH in to run Sudo Raspi-Config. Happy to provide additional info if requested. ThankYou for continuing OctoPi development.

foosel, 2022-03-17:
> Still unable to SSH in to run Sudo Raspi-Config.

That sounds more like an issue of Octo**PI** (the underlying OS), not Octo**PRINT** (the print server that runs on it and for which this release candidate is for). Best get in touch about those SSH issues on the [forums](https://community.octoprint.org) or [discord](https://discord.octoprint.org).

Cpat39, 2022-03-17:
By underlying OS I assume you mean MacOS. I have published for 1.8 1.7.3 on Forum a couple of weeks ago and no discussion of about OS occurred. I’ll contact Apple if you ccnsider it to be a MacOS issue.

foosel, 2022-03-17:
No, I mean Octo**PI**, aka the image you probably flashed to an SD card to have OctoPrint run on your Pi. If you can't ssh into it, writer it's your Mac, or the Pi, or your network, but it isn't OctoPrint since OctoPrint doesn't even know what ssh is.

https://community.octoprint.org/t/what-is-the-difference-between-octoprint-and-octopi-are-they-the-same-thing/185

Cpat39, 2022-03-17:
Gina, I have probably flashed new micro SD’s at least 10-12 times and added an empty SSH file. 

Found this on GitHub. Does not specifically address Terminal, but does say Terminal is broken in Monterey. - https://github.com/microsoft/vscode/issues/143524 <https://github.com/microsoft/vscode/issues/143524>

Crowlord, 2022-03-23:
Running on Prusa Mini, OctoPi 0.17.0. 

WiFi Status
UI Customizer
Timelapse Purger
Slicer Thumbnails
Resource Monitor
Print Track
OctoVox Plugin
Network Health Monitor
M117 Speech Synthesis"
Exclude Region
DeleteAfterPrint Plugin
Autoselect Plugin
DisplayLayerProgress Plugin

No issues noted.

JohnOCFII, 2022-03-24:
OctoPrint 1.8.0rc2 seems to be working well for me:

Uploaded new GCODE to OctoPrint
Printed GCODE on Prusa MK3S+/wMMU2+ in single extruder mode
GCODE viewer worked well
Terminal worked well
Timelapse worked well
The Prusa MMU2 Select Filament plugin worked well
OctoSlack plugin worked well
Print Job History plugin worked well
Telegram plugin worked well

avandeputte, 2022-03-28:
After upgrade from stable to rc3 from the UI,  octopi would crash.

The error is below.  I rolled back to stable and all is well now.
```
ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security' (/home/pi/oprint/lib/python3.7/site-packages/werkzeug/security.py)
2022-03-28 15:32:16,105 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 16 mixin implementations, 30 hook handlers
2022-03-28 15:32:16,121 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 8, in <module>
    sys.exit(main())
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/__init__.py", line 936, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/cli/server.py", line 330, in serve_command
    disable_color=no_color,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/cli/server.py", line 145, in run_server
    from octoprint.server import CannotStartServerException, Server
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/__init__.py", line 33, in <module>
    from flask_login import (  # noqa: F401
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_login/__init__.py", line 16, in <module>
    from .login_manager import LoginManager
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_login/login_manager.py", line 24, in <module>
    from .utils import (login_url as make_login_url, _create_identifier,
  File "/home/pi/oprint/lib/python3.7/site-packages/flask_login/utils.py", line 13, in <module>
    from werkzeug.security import safe_str_cmp
ImportError: cannot import name 'safe_str_cmp' from 'werkzeug.security' (/home/pi/oprint/lib/python3.7/site-packages/werkzeug/security.py)
```

b-morgan, 2022-03-28:
When I click "Reload Now" on 1.8.0rc2 I get a "500 Internal Server Error" instead of the expected "Connecting..." and "Reloading..." followed by the actual OctoPrint UI. Clicking reload on the browser (Chrome on Windows 10) and I get the expected two messages before the UI loads. I don't remember seeing this on 1.7.3 (and earlier). Here is the bundle and I'm off to open a bug report.

[octoprint-systeminfo-20220328144354.zip](https://github.com/OctoPrint/OctoPrint/files/8366201/octoprint-systeminfo-20220328144354.zip)

cp2004, 2022-03-28:
@b-morgan I can see an entry describing that issue in the draft changelog for 1.8.0rc3, so it's possible Gina's already found the problem and fixed it. However, I don't currently see a commit relating to that change on the branch.

foosel, 2022-03-29:
Yes, I stumbled across this yesterday myself while running the first set of update tests for rc3 and fixed it but didn't yet push the fix, only updated the changelog.

@avandeputte 

> After upgrade from stable to rc3 from the UI, octopi would crash.

Stable = what version? 1.8.0rc3 isn't released yet as of right now, you probably meant 1.8.0rc2? And this doesn't look like Octo**PI** did crash but rather Octo**PRINT** ([please don't confuse the two](http://faq.octoprint.org/octoprint-vs-octopi)), and due to a corrupted install.

If this is reproducible on updates then please open a *full bug report*, but I doubt I can fix this in any way on my end, if files are missing from the file system then they are missing.

foosel, 2022-03-29:
Hah, correction, I just got the same error when running the unit tests in CI. Apparently there was an update of a third party dependency that broke things.

foosel, 2022-03-29:
1.8.0rc3 has been released, new feedback ticket is #4475

closed 2022-03-29 by foosel