OctoPrint/OctoPrint #4459: Gcode rendering not working properly in 1.8.0 RC1

mod38, 2022-03-15:
### The problem

After upgrading to 1.8.0 RC1, the gcode viewer doesn't work properly. At first I thought it was freezing up but when  observed it for some time I saw that the progress bar below the viewport was moving too quickly and when it reached 100% the dispay stopped updating until the start of the next layer. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0 RC1

### Operating system running OctoPrint

OctoPi 1.18.0

### Printer model & used firmware incl. version

Prusa Mini

### Browser and version of browser, operating system running browser

Firefox and Chrome on Windows 10 PC but also Chrome on a Raspberry Pi 4B and on a Samsung Tab2S Android tablet

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220315135122.zip](https://github.com/OctoPrint/OctoPrint/files/8253894/octoprint-systeminfo-20220315135122.zip)

I also downgraded to 1.7.3 to check and there was no problem with that version.

foosel, 2022-03-15:
@mod38 could you also provide a gcode file with which you can reproduce this? I cannot currently reproduce this and would like to make sure I have at least a file which definitely causes problems on your end.

mod38, 2022-03-15:
[Gear_bookmark_0.15mm_PLA_MINI_10m.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/8254327/Gear_bookmark_0.15mm_PLA_MINI_10m.gcode.zip)
(I had to zip as gcode files are not accepted
[image]
)

foosel, 2022-03-15:
Yeah, good thing I asked for that file, definitely something weird going on in there and my local test files are working just as expected. Thanks!

foosel, 2022-03-15:
Nope, something about how this file is sliced just unearthed a bug in how the parsed data structure the gcode viewer creates is handed over from a worker to the actual UI. Half the layer contents never arrive. Currently trying to figure out why

foosel, 2022-03-15:
I think I finally got it, at least I can no longer reproduce and other test files are still working as expected as well. Fix ready for 1.8.0rc2

mod38, 2022-03-15:
Great! Sorry to be trouble/ glad to be of assistance!

foosel, 2022-03-15:
Absolutely no trouble (well, apart from the initial reluctance to open a ticket 😉)! The RC phase is meant to unearth stuff like this, so I'm glad your files turned out to be structured in such a way that this precise bug was triggered and thus could be hunted down BEFORE this went stable, even though the bug itself caused me some more grey hair 😂

foosel, 2022-03-16:
1.8.0rc2 has been released.

commit 33cef575 (2022-03-15): 🐛 GCV: Fix layer transmission logic

Only ever send *completed* layers to the frontend. Otherwise in case of
zHops and such we risk sending incomplete layers and missing whole
command segments in the process.

Closes #4459
closed 2022-03-16 by foosel