OctoPrint/OctoPrint #4460: [Request][1.8.0rc1] Expose `serial.ignoreEmptyPorts` in UI

foosel, 2022-03-15:
### Is your feature request related to a problem? Please describe.

```
No serial port found, are you sure your printer is physically connected and supported? Try [refreshing] and if that doesn't help please see [the FAQ]
```

before: I could switch on the printer (from the menu, by GPIO)  - then just click "connect" 
Now: I see the big red text as if something was wrong, need to click "refresh"  - THEN click "connect"

To mitigate: 
-have an option to auto-connect whenever a serial device spawns, (switch on power .. it connects)
-have an option to disable this new "feature" 

_Originally posted by @AndKe in https://github.com/OctoPrint/OctoPrint/issues/4452#issuecomment-1068299388_

### Describe the solution you'd like

`serial.ignoreEmptyPorts` already exists and is exposed to the frontend. Make it available in the UI as advanced options for the general serial connection settings.

foosel, 2022-03-15:
Done and ready for 1.8.0rc2

AndKe, 2022-03-15:
Thank you - this is a perfectly fine solution.

foosel, 2022-03-16:
1.8.0rc2 has been released.

commit 6fc2edcc (2022-03-15): 🚸 Expose serial.ignoreEmptyPorts on UI

Closes #4460
closed 2022-03-16 by foosel