OctoPrint/OctoPrint #4461: MetadataAnalysisFinished - Return json object instead of python object.

RonLight, 2022-03-15:
### Is your feature request related to a problem? Please describe.

Initially I thought the {result} parameter of MetadataAnalysisFinished was returning json, but the name are not quoted.

The documentation states that it is a Python object, I'm assuming the format of a python object.
{printingArea: {maxX: 53.387, maxY: 90.363,  etc

JSON standard is to quote the names
{"printingArea": {"maxX": 53.387, "maxY": 90.363,    etc

### Describe the solution you'd like

Simply quoting the names would allow a simple json deserilization.

### Describe alternatives you've considered

The exact same values are also stored on the file info. I am able to get the file info via the api.

cp2004, 2022-03-17:
What context is this in? I'm assuming, but not sure, that you're trying to use the event subscriptions to run a command, not in a plugin?

RonLight, 2022-10-11:
My suggestion does not require a bundle or log file.

RonLight, 2022-10-11:
Correct the MetadataAnalysisFinished
The full command is:
wget --post-data '{result}' -O- "https://
<server>/endpoint?Event={__eventname}&name={name}&path={path}"

Thanks
Ron

foosel, 2022-10-11:
> My suggestion does not require a bundle or log file.

You removed the `[Request]` tag from the title of your issue, and therefore didn't create a feature request but a bug report without following the proper template for that. That's why you got asked by the bot to provide a system info bundle, since that is mandatory for bug reports.

> Simply quoting the names would allow a simple json deserilization.

And possibly break many existing event workflows that depend on what they currently get there. So that's a no go. We could however think about allowing jinja2 templates in the subscriber settings, in that case you could do something like 

``` jinja2
wget --post-data '{{result|json}}' -O- "https://
<server>/endpoint?Event={{__eventname}}&name={{name}}&path={{path}}"
```

Frankly not sure though if that is worth the hassle when a single file event plugin could achieve the same and more. Event subscriptions predate the way more powerful plugin system.