OctoPrint/OctoPrint #4463: Gcode viewer stops updating one layer before finish

mod38, 2022-03-16:
### The problem

I am checking that the gcode viewer is working properly after the bugfix #4459.
I ran my test piece which has 10 layers. When it finished I noticed that it only showed the 9th layer on the display. I ran the print again with the webcam in a second window and saw that the gcode display stopped after 9 payers but the printer correctly went on and printed the last layer before stopping.

[Gear_bookmark_0.15mm_PLA_MINI_10m.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/8263252/Gear_bookmark_0.15mm_PLA_MINI_10m.gcode.zip)
[image]
 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.8.0rc2

### Operating system running OctoPrint

OctpPi 0.18.0

### Printer model & used firmware incl. version

Prusa Mini

### Browser and version of browser, operating system running browser

Firefox and Chrome on Windows 10 PC but also Chrome on a Raspberry Pi 4B and on a Samsung Tab2S Android tablet

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220316144851.zip](https://github.com/OctoPrint/OctoPrint/files/8263380/octoprint-systeminfo-20220316144851.zip)

foosel, 2022-03-16:
I already have an idea as to why, I think the logic in the fix for #4459 still has a small fault affecting exactly the final layer of the model.

mod38, 2022-03-16:
I did actually check that it wasn't starting with layer 0 instead of 1 (Python indexing)!

foosel, 2022-03-16:
Nah, I think it is related at the change I made to when the layers are handed over from the parser worker to the UI. It now only happens when the active layer changes, but for the last layer it doesn't change anymore, it just stops processing, and that also needs to trigger a layer transfer, but now no longer does. I didn't notice in my testing, but looking at the code it makes total sense. Should be an easy fix that I'll get to once I'm done with the current task, but I won't release 1.8.0rc3 just for that and rather see what else comes up in the next week or so. Should this one stay the only one found, it of course will get another RC just for itself ;)

mod38, 2022-03-16:
Great. I am building up quite a stack of my gear bookmarks but fortunately they are very useful.
I'll just add my thanks for all your work.

foosel, 2022-03-16:
You can also always enable the virtual printer, #4459 and my guess is also this one happily reproduce against that as well.

mod38, 2022-03-16:
I never had occasion to use the virtual printer before, it certainly is a useful tool.
As you expected, the behavious is the same, ending one layer short.

foosel, 2022-03-17:
Fix ready for 1.8.0rc3

foosel, 2022-03-29:
1.8.0rc3 has been released.

commit 8c9536f3 (2022-03-17): 🐛 GCV: Send the final layer to the UI as well

Closes #4463
closed 2022-03-29 by foosel