OctoPrint/OctoPrint #4464: Unable to navigate folder in ssh. "No such file or directory"

WimVDK, 2022-03-16:
### The problem

Whenever i try a new filament, i first of all do 2 things.

1. Print a temp tower
2. Print a filament swatch with the desired temperature. 

I store these gcode files in directories, and the problem lies with the swatch foldername. I organise these by temperature 195°/200°.... I do not know when this changed but in my current OP when i create a folder like "215°" through the WebGUI it actually gets stored as "215°" at the moment. In a previous version of OP (my current one will be listed under "Version of OctoPrint") the WebUI (or underlying code?) renamed this on the filesystem as "215deg".

Something changed and it's no longer does this, this causes the error (in SmartTTY my ssh client) 
> cd: $'/home/pi/.octoprint/uploads/Filament_Related/SwatchBox/Swatches/210\260': No such file or directory
to display and the folder cannot navigated in ssh

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

not a plugin issue

### Version of OctoPrint

1.7.3

### Operating system running OctoPrint

Octopi 0.18 

### Printer model & used firmware incl. version

Not applicable but CR-10 max, Marlin 2.0.9.2

### Browser and version of browser, operating system running browser

Firefox 98.0.1

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]
[octoprint-systeminfo-20220316204803.zip](https://github.com/OctoPrint/OctoPrint/files/8270801/octoprint-systeminfo-20220316204803.zip)

cp2004, 2022-03-16:
I just tried this in the default windows SSH client (running `ssh pi@octopi.local`) and it seems to be able to handle the degree symbol just fine:

```
(oprint) pi@octopi:~/.octoprint $ cd uploads/
(oprint) pi@octopi:~/.octoprint/uploads $ ls
100°  xyzCalibration_cube.gcode
(oprint) pi@octopi:~/.octoprint/uploads $ cd 100°
(oprint) pi@octopi:~/.octoprint/uploads/100° $ ls
CCR6SE_Rodeclip_v1.gcode
(oprint) pi@octopi:~/.octoprint/uploads/100° $
```

I don't think this is a bug in OctoPrint, if you can't type a degree symbol then that's the SSH client's issue?

If you want to go back to the old behaviour of replacing all the special characters & spaces you can set `server.enforceReallyUniversalFilenames` to true in config.yaml

```
server:
  enforceReallyUniversalFilenames: true
```
(find the relevant section)

*(Just noted)* it seems that's not documented on docs.octoprint.org, so I'll add it

closed 2022-03-16 by cp2004

WimVDK, 2022-03-16:
Oh I don't want to go back to the old method, i like the new one. Seems to work fine in the browser/android apps. So you may be right that it's not an OP issue but an ssh one. I've tried a few ssh clients and SmarTTY was the first i really liked.

I'm a Linux noob so i don't really need much in a client. I'd be happy with being able to store IP/server configurations and predictive command text. Maybe a nice GUI for navigation like SmarTTY has but that's about it really.

Suggestions are welcome

cp2004, 2022-03-16:
If you need suggestions from the community, I suggest getting in touch on the [discord server](https://discord.octoprint.org) or the [community forum](https://community.octoprint.org), since those places are frequented by many whereas the issues here are only bug reports/feature requests for OctoPrint and (very) small number of people actually read these.