OctoPrint/OctoPrint #4471: GCode Viewer mis-representing non-extrusion G2/G3 moves

CmdrCody51, 2022-03-21:
### The problem

ArcWelder will convert travel moves to G2/G3 arcs (it even converts wipes to arc).
GCode viewer ignores G3/G3 with no extrusion and generates chord rather than arc.
[image]
[TubeLevelerBase1.zip](https://github.com/OctoPrint/OctoPrint/files/8319736/TubeLevelerBase1.zip)
[octoprint-systeminfo-20220321182011.zip](https://github.com/OctoPrint/OctoPrint/files/8319748/octoprint-systeminfo-20220321182011.zip)

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Viewer is a plugin

### Version of OctoPrint

1.8.0rc2

### Operating system running OctoPrint

Raspberry Bullseye

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

Firefoc 98.01 64-bit

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2022-03-22:
Seems to also happen in 1.7.3 as far as I can see, so this isn't a regression in 1.8.0rc and will have to wait until 1.9.0.

> ### If you could not test in safe mode, please state why
> 
> Viewer is a plugin

For the record, while the viewer is a plugin, it is a *bundled* plugin, and those stay enabled even in safe mode. Safe mode only disables third party plugins and language packs, anything that comes batteries included with OctoPrint is still there in safe mode.

CmdrCody51, 2022-03-22:
Oh absolutely probably is not a 'new' bug - I just happened to update both OctoPrint and then ArcWelder Lib after.
Kinda thought Viewer would be safe for Safe Mode - meh.
If I find it first, I will report back here.

CmdrCody51, 2022-03-22:
That was easy! You write good code! **Against 1.8.0rc2**
renderer.js line 1044-1045:
===============================
                    ctx.lineWidth = lineWidthFactor;
                    ctx.lineTo(x, y);
===============================
replace last line [ctx.lineTo(x, y);] with copy-paste
lines 1052-1065
===============================
                    if (cmd.direction !== undefined && cmd.direction !== 0) {
                        var arc = getArcParams(cmd);
                        var ccw = cmd.direction < 0; // Y-axis is inverted so direction is also inverted

                        if (renderOptions["showDebugArcs"] && !isNotCurrentLayer) {
                            strokePathIfNeeded("debugarc", "#ff0000");
                            drawDebugArc(arc, ccw);
                            strokePathIfNeeded("extrude", getColorLineForTool(tool));
                        }

                        ctx.arc(arc.x, arc.y, arc.r, arc.startAngle, arc.endAngle, ccw);
                    } else {
                        ctx.lineTo(x, y);
                    }
=============================
You can even get it into 1.8.0  8-)

foosel, 2022-03-23:
As mentioned on Discord, fixed yesterday but couldn't push due to GitHub outage, now pushed and will be in 1.9.0 (1.8.0 is frozen now it has entered RC phase and only will get regression fixes and fixes in new functionality). Pretty much the same as yours.

commit 34e935bd (2022-03-23): 🐛 GCV: Moves may be arcs too

Closes #4471
closed 2023-05-23 by foosel