OctoPrint/OctoPrint #4472: Limited network connectivity introduces performance degradation

bzip22, 2022-03-22:
### The problem

It appears when Octoprint is in a disconnected serial state it is entirely functional and speedy while running on a newer model Raspberry pi with Octopi.

However after hitting the 'connect' button to establish a serial connection the service causes Octopi OS to freeze for a few minutes before returning to normal and still resulting a disconnected state (and sometimes not). This impacts navigation of the WebUI along with accessibility via SSH.

In a unique-to-me scenario my home network sits behind a firewall that has some pretty strict traffic filtering. What I have discovered is that clicking "connect" to initiate a serial connection with my printer results in Octopi making some immediate calls to the internet. This traffic would be dropped by the firewall appliance resulting in the described performance degradation.

If I make exceptions to this filtering and retest, the same outbound traffic is attempted but the serial connection is established instantly, resulting in no hang of the service and resulting performance degradation.

Example: Filtering disabled - a packet capture indicates a DNS query to octopodprint.com (52.43.57.186) followed by some non-TLS traffic - tcp port 80 (why no SSL here but SSL with a different FQDN?).

Example: Filtering enabled - a packet capture indicates a DNS query to octoprint.org (159.69.6.198) and then all subsequent traffic to this destination on tcp port 443 is denied.
Additional DNS queries for this particular capture: plugins.octoprint.org

While this probably wouldn't impact the majority of the user base, perhaps this functionality can be improved so that this hang does not occur.

[image]
[image]

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

This seems core functionality related.

### Version of OctoPrint

1.8.0rc2 (and non-rc)

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Ender 5 Pro Merlin 1.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220322012259.zip](https://github.com/OctoPrint/OctoPrint/files/8320141/octoprint-systeminfo-20220322012259.zip)

cp2004, 2022-03-22:
> Did the issue persist even in safe mode?
I cannot test this issue in safe mode (state why below)

> If you could not test in safe mode, please state why
This seems core functionality related.

Please test in safe mode regardless - one of the examples you've given in your post is octopodprint.com, which is a 3rd party plugin/service, not core related at all.

bzip22, 2022-03-22:
> Please test in safe mode regardless - one of the examples you've given in your post is octopodprint.com, which is a 3rd party plugin/service, not core related at all.

well, it certainly would seem that Octopod was the cause of the behavior. no such calls to the internet or service hang happens with the plugin removed or running in safe mode.

i apologize for the false report!

closed 2022-03-24 by bzip22