OctoPrint/OctoPrint #4473: [Request] Add some kind of auto-refresh of the serial ports list to core OctoPrint

foosel, 2022-03-22:
### Is your feature request related to a problem? Please describe.

In 1.8.0 OctoPrint introduces a warning message if no serial ports are currently detected to connect to (see #4237), because we've seen a ton of issues by users using power-only USB cables or trying to connect to unsupported printers and needing guidance on detecting issues like this instead of having to wait for support on the forums or Discord.

The way this warning was implemented in 1.8.0rc1 and 1.8.0rc2 it also caused the "Connect" button to become disable, to force people to realize, hey, something's amiss here, and actually read the warning instead of ignoring it and attempting to connect and then still seeking support when that fails. However, disabling the connect button causes issues with certain workflows where users have their printers powered off and thus unrecognized on serial, and only power them on right before attempting to connect. The auto connect in that cause would then take care of refreshing the ports list, detect the printer and connect just fine. In 1.8.0rc1 and 2 however this required an additional click on refresh before it worked, which means in 1.8.0rc3 the connect button will be re-enabled in these cases.

However, long term it would be the better solution to disable it again if there really are no serial ports recognized by the system and instead have OctoPrint automatically detect new ports, kinda like the PortLister plugin does today, but natively built into core.

### Describe the solution you'd like

OctoPrint being platform agnostic and the PortLister plugin only working on POSIX systems, the same approach like it takes (watchdog & monitoring of all of `/dev` for file additions) won't work. Instead, either a regular auto-refresh (e.g. 1/s) while no ports are detected should be done, or, probably cleaner, a watchdog on the *configured* serial port patterns under POSIX and a regular polling approach for Windows.

### Describe alternatives you've considered

Renaming the connect button to "Refresh & Connect" if no port is detected, and actually having it run a refresh on click in such cases, but that also felt awkward in case of no change in available serial ports after.

foosel, 2022-11-17:
Implemented and ready for 1.9.0.

In the end it became a polling implementation that will be active while there's no connection established to the printer. Every second the current serial post list is fetched, if there is a difference to the prior fetch an event is fired that prompts the frontend to update. Once a connection is started the monitoring stops, to be restarted after the connection is closed. 

Can be disabled via `serial.autorefresh`. The interval can also be changed via `serial.autorefreshInterval`. Currently neither is exposed on the UI as I'm not sure we really need that.

foosel, 2022-11-17:
[image]

jneilliii, 2023-03-31:
Quick question on the implementation. Once the new connection is detected from the previous fetch, if it is a saved port and auto-connect on server startup is enabled does OctoPrint automatically reconnect to the printer the same way that postlister did?

foosel, 2023-05-23:
No, it really just updates the port list. And sorry in the late answer, this slipped through.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel

RobinTail, 2023-08-03:
Thank you for this feature, @foosel 
🙏🏽