OctoPrint/OctoPrint #4474: [Request] Wider left column

N6REJ, 2022-03-27:
### Is your feature request related to a problem? Please describe.

Can't read full filename.

### Describe the solution you'd like

It would be nice if we could have the choice of static vs fluid layout or resizeable left column so that the full filename can be viewed
it appears its currently locked to 8x3 characters which is rather archaic.

### Describe alternatives you've considered

allow for column to be resized on the fly.
show full filename on hover

cp2004, 2022-03-27:
The issue is not the size of the column - long filenames wrap around and are not limited by the width:
 [image]

What you are instead seeing is that filenames that your 3D printer is sending that are stored on its SD card are limited to 8.3 format. That's a limitation of the protocol, if you check the terminal as you connect to the printer you will see how the files are presented. There are options in newer firmware builds to enable long file name support, but these are often disabled by default.

jneilliii, 2022-03-27:
If you do want a wider sidebar, that is possible with ui customizer plugin.

closed 2022-03-27 by cp2004

N6REJ, 2022-03-28:
actually, in my case, they don't wrap at all and are even changed from their actual on card name!
[image]
infact, my icons dont' do anything

jneilliii, 2022-03-28:
Like Charlie already mentioned, these are files stored on your sd card and that's how they are reported to octoprint in 8.3 format.