OctoPrint/OctoPrint #4475: [RC Feedback] Feedback on the 1.8.0rc3 Release Candidate

foosel, 2022-03-28:
# 1.8.0rc3 isn't release yet, this is a placeholder ticket

Please provide *general* feedback on your experience with the 1.8.0rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### ❗👇 IMPORTANT 👇❗
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

b-morgan, 2022-03-29:
Installed successfully. Will do some printing later today.

pingywon, 2022-03-29:
Updated with out any issue. Ran some of my basic tests. All looks good for me so far.

mod38, 2022-03-29:
Updated with no problems and ran simple print correctly.
Printer is Prusa Mini
only self-added plugins are navbar temp and slicer thumbnails.

foosel, 2022-04-05:
1.8.0rc4 has been released, new feedback ticket is #4480

closed 2022-04-05 by foosel