OctoPrint/OctoPrint #4476: 500 Internal Server Error

b-morgan, 2022-03-28:
### The problem

When I click "Restart OctoPrint", and then "Reload Now" on 1.8.0rc2 I get a "500 Internal Server Error" instead of the expected "Connecting..." and "Reloading..." followed by the actual OctoPrint UI. Clicking reload on the browser and I get the expected two messages before the UI loads. I don't remember seeing this on 1.7.3 (and earlier). 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.8.0rc2

### Operating system running OctoPrint

OctoPi 0.18

### Printer model & used firmware incl. version

Recv: FIRMWARE_NAME:Marlin  FIRMWARE_VERSION:2.0.9.0.9 EXTRUDER_TYPE:Universal (Dec  2 2021 09:26:58) SOURCE_CODE_URL:https://gitlab.com/lulzbot3d/marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:LulzBot TAZ 6 EXTRUDER_COUNT:1 UUID:845f003c-aebd-4e53-a6b9-7d0984fde609

### Browser and version of browser, operating system running browser

Firefox 97.0.2 (64-bit) on Windows 10 (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20220328144354.zip](https://github.com/OctoPrint/OctoPrint/files/8366322/octoprint-systeminfo-20220328144354.zip)
[JavaScript Console.txt](https://github.com/OctoPrint/OctoPrint/files/8366370/JavaScript.Console.txt)

foosel, 2022-03-29:
Came across this yesterday during the update tests for rc3 myself and I think I fixed it in 24b5b3a857d0339107904536ba004a64b69920b3

foosel, 2022-03-29:
1.8.0rc3 has been released.

closed 2022-03-29 by foosel