OctoPrint/OctoPrint #4477: flask-login 0.5.x breaks with werkzeug 2.1.x, a solution beyond pinning werkzeug to 2.0.x is needed

foosel, 2022-03-29:
### Problem

Flask-Login 0.5 is no longer compatible to werkzeug 2.1.x+ (see also ccd5fa56a239d42830cb6825b779bc5aa141ee93, which was caused by this, as well as https://github.com/OctoPrint/OctoPrint/issues/4457#issuecomment-1081101141 and maxcountryman/flask-login#636). A newer version isn't released on pypi yet, even though the source repository contains fixes, including for the werkzeug situation. It is unclear if and if so when a version 0.6 of Flask-Login might see an official release (there hasn't been any release activity in close to two years now), and discussion around the topic seems to potentially not be desired by the maintainer(s?) as issues on it get locked.

### Possible solutions

I currently only see two options here really that allow us to update werkzeug to versions beyond 2.0.x...

  1. Vendor bundle flask-login and merge in patches from here on as needed to no longer have to rely on the upstream release cycle. Might cause issues with third party plugins however that expect to be able to import from `flask_login`, if those exist. That needs to be checked.
  2. Monkey patch the problematic removed function that flask-login relies on back into werkzeug. It would be a minimally invasive monkey patch that would basically just set `werkzeug.security.safe_str_cmp` to `hmac.compare_digest`. The upside is that this would keep everything out there working as is, but at the cost of now sitting on yet another monkey patch and the next update of something in the whole flask framework potentially causing the next issue with flask-login. Still, I'm more leaning towards this option at the moment.

Let me just once again voice my dislike for libraries that introduce breaking changes on minor releases. It's annoying AF, especially on third party dependencies of third party dependencies.

cp2004, 2022-04-05:
Flask-login 0.6.0 was released a few days ago. It looks like one of the locked issues on the F-L repo have less savoury original comments (now edited) from people complaining...

There's a few changes between the releases *apart* from compatibility, so I would suggest bumping it and we can give it a test in the maintenance branch.

foosel, 2022-04-06:
Flask-Login has been updated to 0.6 and the version range of werkzeug updated as well. Let the testing begin.

commit 2e4af01b (2022-04-06): ⬆️ Upgrade flask-login & werkzeug

Closes #4477
closed 2023-05-23 by foosel