OctoPrint/OctoPrint #4479: 1.8.0rc3 update ui don't start Sticky Pad(0.1.4) error message

giloser, 2022-04-04:
### The problem

Hi,

After update to the 1.8.0 rc3 for the very first time, the ui seems to start loading and then send an error 500 and ask to check the console and octoprint.log

in the octoprint.log I found the following info

2022-04-04 13:31:54,504 - octoprint.plugins.telegram - DEBUG - self.tmsg = <octoprint_telegram.telegramNotifications.TMSG object at 0xa84d89f0>
2022-04-04 13:31:54,504 - octoprint.plugin - DEBUG - Calling on_event on uicustomizer
2022-04-04 13:31:56,537 - octoprint.server.api.settings - ERROR - Could not load settings for plugin Sticky Pad (0.1.4)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 375, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/types.py", line 1605, in on_settings_load
    data = copy.deepcopy(self._settings.get_all_data(merged=True))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 595, in get_all_data
    return self.settings.get(self._prefix_path(), **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 1953, in get
    result = process()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 1947, in process
    return self._get_value(path, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 1892, in _get_value
    parent_path + [key], only_local=not incl_defaults, merged=merged
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 653, in get_by_path
    {k: v for k, v in current.items() if k.startswith(prefix)}, prefix=prefix
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 590, in _unflatten
    current[subkeys[-1]] = value
TypeError: 'str' object does not support item assignment
2022-04-04 13:31:56,892 - octoprint - ERROR - Exception on /api/settings [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2077, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1241, in decorated_function
    etag = etag_factory(lm)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 61, in _etag
    hash_update(repr(settings().effective))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 1094, in effective
    return self._map.deep_dict()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 611, in deep_dict
    return self._unflatten(self._chainmap)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/settings.py", line 590, in _unflatten
    current[subkeys[-1]] = value
TypeError: 'str' object does not support item assignment

After manual uninstall of the plugin StickyPad the application start correctly again.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

don't know how to start the safe mode from command line as I could not access UI

### Version of OctoPrint

1.8.0rc3

### Operating system running OctoPrint

OctoPi 0.17.0

### Printer model & used firmware incl. version

Ender3 Marlin 2.0.7.2

### Browser and version of browser, operating system running browser

Brave Version 1.37.109 Chromium: 100.0.4896.60 (Official Build) (64-bit),  Windows 10 V21H2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]
[octoprint-systeminfo-20220404140006.zip](https://github.com/OctoPrint/OctoPrint/files/8409478/octoprint-systeminfo-20220404140006.zip)

foosel, 2022-04-04:
Oh dear, looks like another RC might be needed after all, thanks for reporting.

jneilliii, 2022-04-04:
Interesting, never actually looked at how it was being saved from the line [here](https://github.com/jneilliii/OctoPrint-StickyPad/blob/da2ded9d56e0d3cfe797fc45854568c05bd4a4a6/octoprint_stickypad/static/js/stickypad.js#L33), but I could definitely see this being something that would need to be addressed for other plugins too. For example, what happens if you don't define a default setting and it gets added by the plugin's settings interface?

foosel, 2022-04-04:
THAT case is actually covered, it checks for existence of the key and against its value being `None`, but a wrong type check so far was missing. I've added the above case as a unit test and am currently trying to iron it out.

foosel, 2022-04-05:
Hopefully fixed by the above commit, ready for 1.8.0rc4

closed 2022-04-05 by foosel

giloser, 2022-04-05:
It's working fine thanks a lot!

commit f4b4f50c (2022-04-05): 🐛 Fix handling of invalid intermittend values

We not only ignore None values on our subkey paths, we ignore
anything that isn't a dict we can further level down in.

Closes #4479